platformio/platformio-core #3848: Provide support to assembly laungage and C for 8051 MCU family

jayrc2026, 2021-02-15:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

maxgerhardt, 2021-02-15:
This issue is very incomplete without much info.

Also, you can already create PlatformIO projects for 8051 MCUs in C and assembly. See https://github.com/platformio/platform-intel_mcs51. The used assembler is `sdas8051` and the C compiler is `sdcc`.

ivankravets, 2021-02-27:
See https://github.com/platformio/platform-intel_mcs51

closed 2021-02-27 by ivankravets