platformio/platformio-core #3854: Compiling for arduino simylation fails

wogoos, 2021-02-19:
Hi 
I have installed WizIO because I want a tools to test esp/atmega/stm parts of code on my PC without uploading it to the controller all the time. The installation is visible on my latest version of PlatformIO v5.1.0  as WizIO v1.03
I made a very simple sketch basically a Serial.print("hello");
My MinGW path is D:\01_Software\MinGW\bin and added that to the environment variable path.
My sketch   path is D:\Users\wogoos\Doc_Library\Software\_ESP\Projects_ESP8266\Ring_Buffer
I build with [env:emulator] and this is what happens

> Executing task: C:\Users\wogoo\.platformio\penv\Scripts\pio.exe run --environment emulator <
Processing emulator (platform: wizio; board: emulator; framework: arduino)
----------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/Wiz-IO/framework-wizio.git
Error: Traceback (most recent call last):
  File "c:\users\wogoo\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\wogoo\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\wogoo\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\wogoo\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\wogoo\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
.
.
.
en goes on.

Any idea what i'm doing or what goes wrong here.
[PlatformIO dump.txt](https://github.com/platformio/platformio-core/files/6012645/PlatformIO.dump.txt)

valeros, 2021-02-22:
Hi @wogoos ! Do you have Git installed in your system?

ivankravets, 2021-02-27:
Please install https://git-scm.com/downloads and start again.

closed 2021-02-27 by ivankravets

wogoos, 2021-03-02:
Guys I haven't installed it but will thanks for the tip