platformio/platformio-core #3855: .platformio folder not found in C:\Users\MyName thus a lot of errors

Tillema, 2021-02-19:
I'm kinda frustrated. Working on this issue for multiple days now. So, I beg your pardon for being a bit grumpy.

While cleaning my computer, the .platformio folder in C:\Users\simon was accidentally removed. Then the recycle bin was emptied, and I don't have a windows.old folder anymore. So, it lost quite some indispensable stuff. Now my question is: Where to download the .platformio folder?

While hitting Ctrl+Alt+B to build the firmware, it gives me "The terminal process failed to launch: Path to shell executable "c:\Users\simon\Downloads\Marlin-2.0.x\Marlin-2.0.x\pio.exe" does not exist.", So, I went to find the pio.exe file, which appears not to be on my computer since about a week now.

I'm trying to run the last version of platformio extension on the last version of Visual Studio Code. I'm working on Windows 10, 64bit.
Already tried:
- Googled for 3 days, at least 3 hours a day;
- Updating Windows and other relevant stuff;
- Reinstalling VSCode;
- Reinstalling PlatformIO extension;
- Removed Python and installed the latest version;
- Removing all VSCode and PlatformIO related stuff I could find on my PC, including registry cleanup, and then install again.

About a half year ago, I had the same issue. Not able to fix, so resetted my PC back to factory settings. Then it worked, but I am not going to reset my PC this time.

If you need more info, please let me know. I'm not good at describing issues.

I have to compile some 3D printer firmware (namely Marlin) for a friend.

I'm good with computers, but ain't a wizard. Not able to fix this damn issue.

Any help would be greatly appreciated.

Thanks!

Simon

Jason2866, 2021-02-22:
Probably a not up to date platformio config in the project you want to compile.
Link to the .git would help to help you

ivankravets, 2021-02-27:
Please remove `C:\Users\simon\.platformio` and `C:\.platformio` (if exists) folders and restart VSCode.

closed 2021-02-27 by ivankravets