platformio/platformio-core #3856: Cannot unpublish package w/ non "x.y.z" formatted version; error "Invalid simple block"

fxkr, 2021-02-20:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Linux

**PlatformIO Version** (`platformio --version`):

Your issue template is wrong, you should ask for `pio --version`.

```
% pio --version
PlatformIO Core, version 5.1.0
```

### Description of problem

I published a version `1.0.18`, but I needed to redo it, so I unpublished it and published `1.0.18.2`. I need to redo this too, so I want to unpublish it. However, it fails:

```% pio package unpublish 'libsodium-unofficial/libsodium @ 1.0.18.2'                    
Error: Traceback (most recent call last):
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/commands/package.py", line 123, in package_unpublish
    spec = PackageSpec(package)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/package/meta.py", line 124, in __init__
    self._parse(self.raw)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/package/meta.py", line 227, in _parse
    raw = parser(raw)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/package/meta.py", line 250, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/platformio/package/meta.py", line 167, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/semantic_version/base.py", line 618, in __init__
    self.clause = self._parse_to_clause(expression)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/semantic_version/base.py", line 1014, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/semantic_version/base.py", line 1035, in parse
    clause &= cls.parse_block(block)
  File "/home/fxkr/.platformio/penv/lib64/python3.9/site-packages/semantic_version/base.py", line 1059, in parse_block
    raise ValueError("Invalid simple spec component: %r" % expr)
ValueError: Invalid simple spec component: '1.0.18.2'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

...And that's why I'm here. You could say it correctly rejects the invalid input, but note that it has already accepted it when I published the package, so now I'm stuck in this invalid state with a broken package published that I cannot get rid of.

I also cannot unpublish the entire package, even though the docs say it should work:

```
% pio package unpublish 'libsodium-unofficial/libsodium'
Error: Version 'None' not found
```

I also cannot install this package:

```
% pio lib install libsodium-unofficial/libsodium                                                                                                           :(
Library Storage: (.........path removed...........)/.pio/libdeps/native
Library Manager: Installing libsodium-unofficial/libsodium
Error: Could not install package 'libsodium-unofficial/libsodium' for 'linux_x86_64' system
```

Not sure why it says `linux_x86_64`? I did play with that platform earlier, but my only platform right now is `native`. Maybe this is just the message, and the underlying error is not shown here? But I don't know that.

I have tried deleting the `.pio` directory.

How can I make the package gone?

What is the correct way to version a package when you need to re-release, but x.y.z is predetermined because you're packaging somebody elses library? The docs claim platformio does SemVer, but from what I can tell from experiments that is not true at all (PlatformIO seems to do `x.y.z`, SemVer is much more. The `x.y.z.a` format above is not SemVer, but I tried with other version strings that are SemVer compliant like `1.2.3+4` and it also broke in different places).

```
{
  "name": "libsodium",
  "version": "1.0.18.2",
  "description": "A modern, portable, easy to use crypto library. This PlatformIO package is unofficial / not maintained by the libsodium team.",
  "keywords": "crypto, sodium, NaCl",
  "repository":
  {
    "type": "git",
    "url": "https://github.com/jedisct1/libsodium"
  },
  "license": "ISC",
  "homepage": "https://libsodium.org/",
  "dependencies": {
  },
  "frameworks": "*",
  "platforms": "*",
  "build":{
    "includeDir": "src/libsodium/include",
    "flags": [
      "-Isrc/libsodium/include/sodium",
      "-DCONFIGURED=1"
    ]
  }
}
```

ivankravets, 2021-02-27:
The package has been deleted. Please publish it again.

closed 2021-02-27 by ivankravets

fxkr, 2021-02-28:
Thanks, that works for me!

(Even though it does not fix the underlying bug that publish allows version numbers that make other commands fail.)