platformio/platformio-core #3858: 'checkprogsize' failure does not cancel the build right away

mcspr, 2021-02-22:
### Configuration

**Operating system**:

Ubuntu 18.04.5 LTS and Windows 10H2

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 5.1.0

### Description of problem

As noticed in the ESPEasy CI job:
https://travis-ci.org/github/letscontrolit/ESPEasy/jobs/759967019#L1200-L1208

The build goes over the supported 1MB, so checkprogsize calls an `env.Exit(1)`. However, binary is still generated from the .elf and user extra scripts are also called as the result:
https://github.com/platformio/platformio-core/blob/9c20ab81cb68f1ffb7a8cac22ce95c4c797643ec/platformio/builder/tools/platformio.py#L86
https://github.com/platformio/platformio-core/blob/9c20ab81cb68f1ffb7a8cac22ce95c4c797643ec/platformio/builder/tools/pioupload.py#L271

Using a dummy project with a simple `extra_script = script.py` containing `env.AddPreAction("${BUILD_DIR}/${PROGNAME}.bin` somewhere, it will still call the checkprogsize right after, but *will* cancel the .bin generation.

### Expected results

.bin file is not generated when checkprogsize fails, and user scripts are not called

Just like this works:
```python
Import("env")
def stop(source, target, env):
    env.Exit(1)
env.AddPreAction("${BUILD_DIR}/${PROGNAME}.bin", stop)
```
It will stop the build. However, it will not stop checkprogsize function from being called even if it comes before it (based on order in the log)

### Issue example

I also tried this locally on a Windows machine:

platformio.ini
```ini
[env]
platform = espressif8266
board = d1_mini
framework = arduino

[env:fail]
```

modified platformio/builder/tools/pioupload.py, so we can reliably fail
(`board_upload.maximum_size = 0` does not seem to work :/)
```diff
diff --git a/platformio/builder/tools/pioupload.py b/platformio/builder/tools/pioupload.py
index 5e1117b5..71573fdd 100644
--- a/platformio/builder/tools/pioupload.py
+++ b/platformio/builder/tools/pioupload.py
@@ -263,7 +263,7 @@ def CheckUploadSize(_, target, source, env):
     #         "Error: The data size (%d bytes) is greater "
     #         "than maximum allowed (%s bytes)\n" % (data_size, data_max_size))
     #     env.Exit(1)
-    if program_size > program_max_size:
+    if True:
         sys.stderr.write(
             "Error: The program size (%d bytes) is greater "
             "than maximum allowed (%s bytes)\n" % (program_size, program_max_size)
```

log
```
> pio run -e fail
...
Linking .pio\build\fail\firmware.elf
Retrieving maximum program size .pio\build\fail\firmware.elf
Checking size .pio\build\fail\firmware.elf
Building .pio\build\fail\firmware.bin
Creating BIN file ".pio\build\fail\firmware.bin" using "C:\Users\maxim\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\fail\firmware.elf"
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  32.9% (used 26976 bytes from 81920 bytes)
Error: The program size (266316 bytes) is greater than maximum allowed (1044464 bytes)
Flash: [===       ]  25.5% (used 266316 bytes from 1044464 bytes)
*** [checkprogsize] Explicit exit, status 1
======================================================================== [FAILED] Took 9.00 seconds ========================================================================
(
```

As seem above, firmware.bin file is created

mcspr, 2021-02-22:
I also noticed the dependency on the 'program' here:
https://github.com/platformio/platformio-core/blob/9c20ab81cb68f1ffb7a8cac22ce95c4c797643ec/platformio/builder/main.py#L185

So, I was able to do this to fix my issue:
```diff
diff --git a/platformio/builder/tools/platformio.py b/platformio/builder/tools/platformio.py
index 4e1ca9ca..ed144bc2 100644
--- a/platformio/builder/tools/platformio.py
+++ b/platformio/builder/tools/platformio.py
@@ -86,6 +86,7 @@ def BuildProgram(env):
             env.VerboseAction(env.CheckUploadSize, "Checking size $PIOMAINPROG"),
         )
     )
+    env.Depends("${BUILD_DIR}/${PROGNAME}.bin", env.Alias("checkprogsize"))

     print("Building in %s mode" % env.GetBuildType())

```
(assuming, 'program' is .elf and not the .bin, it would seem to be the correct dependency?)

closed 2021-03-08 by valeros

valeros, 2021-03-08:
Hi @mcspr ! Thanks for reporting, should be fixed in the dev branch of the `espressif8266` platform. You can try it out using the [upstream version](https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions) of the platform.