platformio/platformio-core #3862: Project Configuration does not preserve comments in `platformio.ini`

pfeerick, 2021-02-26:
Behaviour Change Request

I think this may go in the 'too hard basket' ... but still have to ask ;)

The (fantastic, btw!) project configuration GUI does some... Makes some questionable changes to the `platformio.ini` configuration file. I shorten the header text on a lot of mine, and also comment my libdeps lines, but by ... 

1. opening the configurator
2. adding a description
3. hitting save 

... this removes the comments, and re-adds the full header to the file. Also, as noted by another user, it sometimes adds 

```
lib_archive = yes
test_build_project_src = no
```
on it's own, even though these options haven't been touched. 

So is it possible to *not* change lines if they haven't been modified... i.e. don't strip the comments, and perhaps not make changes to the header. 

[image]

nickyonge, 2024-03-12:
Just ran into this issue myself. I see the discussion above but this issue is still listed as open. Is PIO going to preserve platformio.ini comments by default? That'd be amazing, but otherwise, I'd suggest closing this issue with Won't Fix

pfeerick, 2024-03-13:
Given the suggestion that `custom_` prefix could be used to embed comments, I agree... either try the suggested ConfigParser which *seems* like it would allow for comments and other formatting with, or just close as a known issue / wont fix and be done with it.

ivankravets, 2024-03-13:
We will assess this package in the upcoming PlatformIO Core 7.0. The last time we tried it (Python 2.7), it was buggy and didn't parse complex INI files very well.