platformio/platformio-core #3864: Debugger startup crash with blackmagic probe as debugger

optorres, 2021-02-28:
### Configuration

**Operating system**: Linux

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.0

### Description of problem

When launching the debugger I get the following error:
```
Error: Traceback (most recent call last):
  File "~/.platformio/penv/lib/python3.9/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "~/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "~/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "~/.platformio/penv/lib/python3.9/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "~/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "~/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "~/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "~/.platformio/penv/lib/python3.9/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "~/.platformio/penv/lib/python3.9/site-packages/platformio/commands/debug/command.py", line 117, in cli
    debug_options = platform.configure_debug_options(debug_options, ide_data)
  File "~/.platformio/platforms/ststm32/platform.py", line 185, in configure_debug_options
    server_executable = debug_options["server"]["executable"].lower()
TypeError: 'NoneType' object is not subscriptable
```

#### Steps to Reproduce

Start a debugging session with `blackmagic` configured as the debug tool.  The debugger crashes instead of starting.

### Example config

Config needs to set `blackmagic` as the debug tool, the rest shouldn't matter.
```ini
[env:mcu]
platform = ststm32
board = ...
framework = ...

upload_protocol = blackmagic
debug_tool = blackmagic
```

Source doesn't matter, anything that compiles should work.

### Additional info

The platform config function assumes that there will be a server, but this is not the case for the black magic probe since gdb connects directly to the serial port that the probe exposes.

I hit this issue with the `ststm32` platform, but I took a quick look and it seems like the other platforms have the same issue so I figured this was probably the best place to report it.

I fixed the issue by moving the assignment to `server_executable` down two lines into the conditional below, since it isn't actually used before then.

valeros, 2021-03-02:
Many thanks for reporting, should be fixed in all affected platforms, please retest with the upstream version of the `ststm32` platform:

```ini
[env:mcu]
platform = https://github.com/platformio/platform-ststm32.git
board = ...
framework = ...

upload_protocol = blackmagic
debug_tool = blackmagic
```

closed 2021-03-02 by valeros

optorres, 2021-03-03:
That works, thanks!

Note that your change adds a default for `executable` instead of adding a default for `server`, so it can still crash if you specify `debug_speed = ...` so that it enters the code block, but I don't think that option is applicable to `blackmagic`, so that's not really an issue.