platformio/platformio-core #3865: Remote agent on Raspbian

lkankowski, 2021-02-28:
Hi

------------------------------------------------------------------

### Configuration

**Operating system**: Raspbian GNU/Linux 10 (buster) [fully updated]

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.0

### Description of problem

It probably wont be the most descriptive report, but I just finished few hours fight with 'pio remote agent start' :)
Finally, after dozen times of reinstalling Platformio, installing pyenv and python 3.8.8, some additional packages (from bash history: cryptography==2.9, pyopenssl, cffi, pyopenssl) and lots of frustration with installing correct version of 'twisted' (finally I couldn't install other than 21.2) I did two successful steps:
1. change file `platformio-core/platformio/package/manager/core.py`:
`    twisted_version = "19.10.0" if PY2 else "20.3.0"
    result = [
        "twisted >= %s" % twisted_version,
    ]`
('twisted ==...' into 'twisted >=...')
2. run command: `CRYPTOGRAPHY_DONT_BUILD_RUST=1 pio remote agent`

and it built.

#### Steps to Reproduce

as above

### Actual Results

### Expected Results

### Additional info
YOU DO GREAT JOB WITH PLATFORMIO!

ivankravets, 2021-03-04:
You just need these packages in your OS 
```
sudo apt install python3-dev libffi-dev
```

The only you start `pio remote agent start`, PlatformIO Core will build dependencies on your machine.

lkankowski, 2021-03-06:
I've tried on almost clean RPI4, and doesn't work :(

```

root@rpi4:/home/pi# apt update && apt -y upgrade
...

root@rpi4:/home/pi# apt install python3-dev libffi-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dh-python libexpat1-dev libpython3-dev libpython3.7-dev python3-distutils python3-lib2to3 python3.7-dev
The following NEW packages will be installed:
  dh-python libexpat1-dev libffi-dev libpython3-dev libpython3.7-dev python3-dev python3-distutils python3-lib2t
  python3.7-dev
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 48.3 MB of archives.
After this operation, 74.1 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf python3-lib2to3 all 3.7.7 kB]
Get:2 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf python3-distutils all 3.42 kB]
Get:3 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf dh-python all 3.20190308B]
Get:4 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libexpat1-dev armhf 2.2.0u1 [127 kB]
Get:5 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libffi-dev armhf 3.2.1-9]
Get:6 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libpython3.7-dev armhf 3eb10u2 [47.2 MB]
Get:9 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3-dev armhf 3.7.3-1 [1,264 B]
Get:7 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libpython3-dev armhf 3.7.1 kB]
Get:8 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf python3.7-dev armhf 3.7.0u2 [522 kB]
Fetched 48.3 MB in 1min 45s (460 kB/s)
Selecting previously unselected package python3-lib2to3.
(Reading database ... 41990 files and directories currently installed.)
Preparing to unpack .../0-python3-lib2to3_3.7.3-1_all.deb ...
Unpacking python3-lib2to3 (3.7.3-1) ...
Selecting previously unselected package python3-distutils.
Preparing to unpack .../1-python3-distutils_3.7.3-1_all.deb ...
Unpacking python3-distutils (3.7.3-1) ...
Selecting previously unselected package dh-python.
Preparing to unpack .../2-dh-python_3.20190308_all.deb ...
Unpacking dh-python (3.20190308) ...
Selecting previously unselected package libexpat1-dev:armhf.
Preparing to unpack .../3-libexpat1-dev_2.2.6-2+deb10u1_armhf.deb ...
Unpacking libexpat1-dev:armhf (2.2.6-2+deb10u1) ...
Selecting previously unselected package libffi-dev:armhf.
Preparing to unpack .../4-libffi-dev_3.2.1-9_armhf.deb ...
Unpacking libffi-dev:armhf (3.2.1-9) ...
Selecting previously unselected package libpython3.7-dev:armhf.
Preparing to unpack .../5-libpython3.7-dev_3.7.3-2+deb10u2_armhf.deb ...
Unpacking libpython3.7-dev:armhf (3.7.3-2+deb10u2) ...
Selecting previously unselected package libpython3-dev:armhf.
Preparing to unpack .../6-libpython3-dev_3.7.3-1_armhf.deb ...
Unpacking libpython3-dev:armhf (3.7.3-1) ...
Selecting previously unselected package python3.7-dev.
Preparing to unpack .../7-python3.7-dev_3.7.3-2+deb10u2_armhf.deb ...
Unpacking python3.7-dev (3.7.3-2+deb10u2) ...
Selecting previously unselected package python3-dev.
Preparing to unpack .../8-python3-dev_3.7.3-1_armhf.deb ...
Unpacking python3-dev (3.7.3-1) ...
Setting up libffi-dev:armhf (3.2.1-9) ...
Setting up libexpat1-dev:armhf (2.2.6-2+deb10u1) ...
Setting up python3-lib2to3 (3.7.3-1) ...
Setting up python3-distutils (3.7.3-1) ...
Setting up dh-python (3.20190308) ...
Setting up libpython3.7-dev:armhf (3.7.3-2+deb10u2) ...
Setting up python3.7-dev (3.7.3-2+deb10u2) ...
Setting up libpython3-dev:armhf (3.7.3-1) ...
Setting up python3-dev (3.7.3-1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...

[... 416 lines cut ...]
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 129, in inject_contrib_pysite
    build_contrib_pysite_package(contrib_pysite_dir)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 153, in build_contrib_pysite_package
    subprocess.check_call(args + [dep])
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.platformio/penv/bin/python', '-m', 'pip', 'install', '--no-compile', '-t', '/home/pi/.platformio/packages/contrib-pysite', '--no-binary', ':all:', 'pyopenssl >= 16.0.0']' returned non-zero exit status 1.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

pi@rpi4:~ $ 
```

ivankravets, 2021-03-14:
It seems the `cryptography` package changed his installation process. See https://cryptography.io/en/latest/faq.html#installing-cryptography-fails-with-error-can-not-find-rust-compiler

Could you try to install `apt install rustup`?

lkankowski, 2021-03-14:
There is no "rustup" in "Raspbian GNU/Linux 10 (buster)" repositories.
Sometimes, but not always, if I run for first time:
`CRYPTOGRAPHY_DONT_BUILD_RUST=1 pio remote agent start`
then compilation finishes with errors (unfortunately I didn't capture them).
And then second run `pio remote agent start` works.

ivankravets, 2021-03-15:
Could you run `pio upgrade --dev`? Does it work now? You don't need to add manually `CRYPTOGRAPHY_DONT_BUILD_RUST`, it's already in the code.

lkankowski, 2021-03-15:
Unfortunately, following error:
```
pi@rpi4:~ $ pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.1.1b1
Release notes: https://docs.platformio.org/en/latest/history.html
pi@rpi4:~ $ pio remote agent start
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 5.1.1b1!

************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
************************************************************************************************************************

Tool Manager: Installing platformio/contrib-pysite @ ~2.37.0
Unpacking  [####################################]  100%
Tool Manager: contrib-pysite @ 2.37.191020 has been installed!
Collecting twisted==20.3.0
  Using cached Twisted-20.3.0.tar.bz2 (3.1 MB)
Collecting zope.interface>=4.4.2
  Using cached zope.interface-5.2.0.tar.gz (227 kB)
Collecting constantly>=15.1
  Using cached constantly-15.1.0.tar.gz (21 kB)
Collecting incremental>=16.10.1
  Using cached incremental-21.3.0.tar.gz (17 kB)
Collecting Automat>=0.3.0
  Using cached Automat-20.2.0.tar.gz (61 kB)
Collecting hyperlink>=17.1.1
  Using cached hyperlink-21.0.0.tar.gz (140 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting PyHamcrest!=1.10.0,>=1.9.0
  Using cached PyHamcrest-2.0.2.tar.gz (44 kB)
Collecting attrs>=19.2.0
  Using cached attrs-20.3.0.tar.gz (164 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting six
  Using cached six-1.15.0.tar.gz (33 kB)
Collecting idna>=2.5
  Using cached idna-3.1.tar.gz (181 kB)
Collecting setuptools
  Using cached setuptools-54.1.2.tar.gz (2.1 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Skipping wheel build for twisted, due to binaries being disabled for it.
Skipping wheel build for Automat, due to binaries being disabled for it.
Skipping wheel build for constantly, due to binaries being disabled for it.
Skipping wheel build for idna, due to binaries being disabled for it.
Skipping wheel build for incremental, due to binaries being disabled for it.
Skipping wheel build for PyHamcrest, due to binaries being disabled for it.
Skipping wheel build for zope.interface, due to binaries being disabled for it.
Skipping wheel build for six, due to binaries being disabled for it.
Building wheels for collected packages: attrs, hyperlink, setuptools
  Building wheel for attrs (PEP 517) ... done
  Created wheel for attrs: filename=attrs-20.3.0-py2.py3-none-any.whl size=49336 sha256=7a21bd44c8b0d2945ee404114c36082f24c38899551fb0485a68345fee769366
  Stored in directory: /home/pi/.cache/pip/wheels/a3/97/8f/66c1f6349d2edefbcedb90cb11548b201eff59c0d74c474ad9
  Building wheel for hyperlink (PEP 517) ... done
  Created wheel for hyperlink: filename=hyperlink-21.0.0-py2.py3-none-any.whl size=74638 sha256=8e9f63a3e361ffed2c690a1c261ea55afd183633c70cd7dbd7fb0892090ea1e5
  Stored in directory: /home/pi/.cache/pip/wheels/88/de/19/cb72b4d9352212aa72590f091be9d29c2ae597def937977879
  Building wheel for setuptools (PEP 517) ... done
  Created wheel for setuptools: filename=setuptools-54.1.2-py3-none-any.whl size=785007 sha256=64e532302168e92c49a27403ef3401d6dc1eb75bad2f56a014500cde7ae505f7
  Stored in directory: /home/pi/.cache/pip/wheels/77/6c/9c/a932d7b60e7bce4716cd528fe848cdcb50448006fb87ebb021
Successfully built attrs hyperlink setuptools
[... 229 lines cut ...]
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 129, in inject_contrib_pysite
    build_contrib_pysite_package(contrib_pysite_dir)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 156, in build_contrib_pysite_package
    subprocess.check_call(args + [dep])
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.platformio/penv/bin/python', '-m', 'pip', 'install', '--no-compile', '-t', '/home/pi/.platformio/packages/contrib-pysite', '--no-binary', ':all:', 'pyopenssl >= 16.0.0']' returned non-zero exit status 1.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
Same error after installing Rust compiler with command `curl https://sh.rustup.rs -sSf | sh` and with command `CRYPTOGRAPHY_DONT_BUILD_RUST=1 pio remote agent start`.
I can try this on completly clean Raspbian, but you must wait a few days.

lkankowski, 2021-03-15:
Completly clean Raspberry OS instalation:
```
Using username "pi".
Server refused our key
pi@192.168.1.112's password:
Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ sudo bash
root@raspberrypi:/home/pi# apt update && apt upgrade
........
done.

root@raspberrypi:/home/pi/.ssh# exit
exit

pi@raspberrypi:~ $ sudo apt install python3-dev libffi-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dh-python libexpat1-dev libpython3-dev libpython3.7-dev python3-distutils python3-lib2to3 python3.7-dev
The following NEW packages will be installed:
  dh-python libexpat1-dev libffi-dev libpython3-dev libpython3.7-dev python3-dev python3-distutils python3-lib2to3 python3.7-dev
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 48.3 MB of archives.
After this operation, 74.1 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf python3-lib2to3 all 3.7.3-1 [76.7 kB]
Get:2 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf python3-distutils all 3.7.3-1 [142 kB]
Get:3 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf dh-python all 3.20190308 [99.3 kB]
Get:4 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libexpat1-dev armhf 2.2.6-2+deb10u1 [127 kB]
Get:5 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libffi-dev armhf 3.2.1-9 [159 kB]
Get:9 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3-dev armhf 3.7.3-1 [1,264 B]
Get:6 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libpython3.7-dev armhf 3.7.3-2+deb10u2 [47.2 MB]
Get:7 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf libpython3-dev armhf 3.7.3-1 [20.1 kB]
Get:8 http://ftp.icm.edu.pl/pub/Linux/distributions/raspbian/raspbian buster/main armhf python3.7-dev armhf 3.7.3-2+deb10u2 [522 kB]
Fetched 48.3 MB in 4min 0s (201 kB/s)
Selecting previously unselected package python3-lib2to3.
(Reading database ... 40126 files and directories currently installed.)
Preparing to unpack .../0-python3-lib2to3_3.7.3-1_all.deb ...
Unpacking python3-lib2to3 (3.7.3-1) ...
Selecting previously unselected package python3-distutils.
Preparing to unpack .../1-python3-distutils_3.7.3-1_all.deb ...
Unpacking python3-distutils (3.7.3-1) ...
Selecting previously unselected package dh-python.
Preparing to unpack .../2-dh-python_3.20190308_all.deb ...
Unpacking dh-python (3.20190308) ...
Selecting previously unselected package libexpat1-dev:armhf.
Preparing to unpack .../3-libexpat1-dev_2.2.6-2+deb10u1_armhf.deb ...
Unpacking libexpat1-dev:armhf (2.2.6-2+deb10u1) ...
Selecting previously unselected package libffi-dev:armhf.
Preparing to unpack .../4-libffi-dev_3.2.1-9_armhf.deb ...
Unpacking libffi-dev:armhf (3.2.1-9) ...
Selecting previously unselected package libpython3.7-dev:armhf.
Preparing to unpack .../5-libpython3.7-dev_3.7.3-2+deb10u2_armhf.deb ...
Unpacking libpython3.7-dev:armhf (3.7.3-2+deb10u2) ...
Selecting previously unselected package libpython3-dev:armhf.
Preparing to unpack .../6-libpython3-dev_3.7.3-1_armhf.deb ...
Unpacking libpython3-dev:armhf (3.7.3-1) ...
Selecting previously unselected package python3.7-dev.
[... 745 lines cut ...]
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 129, in inject_contrib_pysite
    build_contrib_pysite_package(contrib_pysite_dir)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 156, in build_contrib_pysite_package
    subprocess.check_call(args + [dep])
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.platformio/penv/bin/python', '-m', 'pip', 'install', '--no-compile', '-t', '/home/pi/.platformio/packages/contrib-pysite', '--no-binary', ':all:', 'pyopenssl >= 16.0.0']' returned non-zero exit status 1.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

pi@raspberrypi:~ $
```

ivankravets, 2021-03-16:
You need to have the next dependent packages:
```
sudo apt install python3-dev libssl-dev libffi-dev
```

Could you try to install the packages above and restart PlatformIO Remote agent?

lkankowski, 2021-03-16:
Unfortunately:
```
pi@raspberrypi:~ $ sudo apt install python3-dev libssl-dev libffi-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libffi-dev is already the newest version (3.2.1-9).
python3-dev is already the newest version (3.7.3-1).
Suggested packages:
  libssl-doc
The following NEW packages will be installed:
  libssl-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,584 kB of archives.
After this operation, 5,938 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libssl-dev armhf 1.1.1d-0+deb10u5+rpt1 [1,584 kB]
Fetched 1,584 kB in 3s (471 kB/s)
Selecting previously unselected package libssl-dev:armhf.
(Reading database ... 40533 files and directories currently installed.)
Preparing to unpack .../libssl-dev_1.1.1d-0+deb10u5+rpt1_armhf.deb ...
Unpacking libssl-dev:armhf (1.1.1d-0+deb10u5+rpt1) ...
Setting up libssl-dev:armhf (1.1.1d-0+deb10u5+rpt1) ...

pi@raspberrypi:~ $ pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.1.1b1
Release notes: https://docs.platformio.org/en/latest/history.html
pi@raspberrypi:~ $ pio remote agent start
Tool Manager: Installing platformio/contrib-pysite @ ~2.37.0
Unpacking  [####################################]  100%
Tool Manager: contrib-pysite @ 2.37.191020 has been installed!
Collecting twisted==20.3.0
  Using cached Twisted-20.3.0.tar.bz2 (3.1 MB)
Collecting zope.interface>=4.4.2
  Using cached zope.interface-5.2.0.tar.gz (227 kB)
Collecting constantly>=15.1
  Using cached constantly-15.1.0.tar.gz (21 kB)
Collecting incremental>=16.10.1
  Using cached incremental-21.3.0.tar.gz (17 kB)
Collecting Automat>=0.3.0
  Using cached Automat-20.2.0.tar.gz (61 kB)
Collecting hyperlink>=17.1.1
  Using cached hyperlink-21.0.0.tar.gz (140 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting PyHamcrest!=1.10.0,>=1.9.0
  Using cached PyHamcrest-2.0.2.tar.gz (44 kB)
Collecting attrs>=19.2.0
  Using cached attrs-20.3.0.tar.gz (164 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting six
  Using cached six-1.15.0.tar.gz (33 kB)
Collecting idna>=2.5
  Using cached idna-3.1.tar.gz (181 kB)
Collecting setuptools
  Using cached setuptools-54.1.2.tar.gz (2.1 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Skipping wheel build for twisted, due to binaries being disabled for it.
Skipping wheel build for Automat, due to binaries being disabled for it.
Skipping wheel build for constantly, due to binaries being disabled for it.
Skipping wheel build for idna, due to binaries being disabled for it.
Skipping wheel build for incremental, due to binaries being disabled for it.
Skipping wheel build for PyHamcrest, due to binaries being disabled for it.
Skipping wheel build for zope.interface, due to binaries being disabled for it.
[... 146 lines cut ...]
  File "/home/pi/.platformio/packages/contrib-pysite/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/pi/.platformio/packages/contrib-pysite/OpenSSL/crypto.py", line 14, in <module>
    from cryptography import utils, x509
  File "/home/pi/.platformio/packages/contrib-pysite/cryptography/x509/__init__.py", line 7, in <module>
    from cryptography.x509.base import (
  File "/home/pi/.platformio/packages/contrib-pysite/cryptography/x509/base.py", line 13, in <module>
    from cryptography.hazmat.backends import _get_backend
ImportError: cannot import name '_get_backend' from 'cryptography.hazmat.backends' (/home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/backends/__init__.py)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

ivankravets, 2021-12-14:
It seems that the issue has been finally fixed. Please re-test with `pio upgrade --dev`.

lkankowski, 2021-12-14:
Well, now I get following error:
```
...
Collecting cryptography<35.0.0,>=3.3
  Using cached cryptography-3.4.8.tar.gz (546 kB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.platformio/penv/bin/python /tmp/pip-standalone-pip-lnbu5lkc/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-0soqgyxd/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.12; platform_python_implementation != '"'"'PyPy'"'"'' 'setuptools-rust>=0.11.4'
       cwd: None
  Complete output (33 lines):
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Using cached setuptools-59.6.0.tar.gz (2.3 MB)
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
    Installing backend dependencies: started
    Installing backend dependencies: finished with status 'done'
    Preparing metadata (pyproject.toml): started
    Preparing metadata (pyproject.toml): finished with status 'done'
  Collecting wheel
    Using cached wheel-0.37.0.tar.gz (65 kB)
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): finished with status 'done'
  Collecting cffi>=1.12
    Using cached cffi-1.15.0.tar.gz (484 kB)
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): finished with status 'done'
  Collecting setuptools-rust>=0.11.4
    Using cached setuptools-rust-1.1.2.tar.gz (285 kB)
    Installing build dependencies: started
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
  ERROR: Some build dependencies for setuptools-rust>=0.11.4 from https://files.pythonhosted.org/packages/67/08/e1aa2c582c62ac76e4d60f8e454bd3bba933781a06a88b4e38797445822a/setuptools-rust-1.1.2.tar.gz#sha256=a0adb9b503c0ffc4e8fe80b7c617898cefa78049983aaaea7f747e153a3e65d1 conflict with the backend dependencies: setuptools_scm==6.0.1 is incompatible with setuptools_scm>=6.3.2.
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/cc/98/8a258ab4787e6f835d350639792527d2eb7946ff9fc0caca9c3f4cf5dcfe/cryptography-3.4.8.tar.gz#sha256=94cc5ed4ceaefcbe5bf38c8fba6a21fc1d365bb8fb826ea1688e3370b2e24a1c (from https://pypi.org/simple/cryptography/) (requires-python:>=3.6). Command errored out with exit status 1: /home/pi/.platformio/penv/bin/python /tmp/pip-standalone-pip-lnbu5lkc/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-0soqgyxd/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.12; platform_python_implementation != '"'"'PyPy'"'"'' 'setuptools-rust>=0.11.4' Check the logs for full command output.
  Using cached cryptography-3.4.7.tar.gz (546 kB)
  Installing build dependencies ... -
```
but I must admit, that the system isn't clean. I've only deleted ~/.platformio, ~/.cache/pip, /tmp and still error as pasted above.

When it fails, it tries another version of `cryptography`: 3.4.8, 3.4.7, 3.4.6, ...
Same on another RPI 4, but without latest system updates. Tomorrow I can try on clean Raspbian OS.

ivankravets, 2021-12-15:
@lkankowski could you remove `~/.platformio/penv` and `~/.platformio/packages/contrib-pysite`? Then , run installer script https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux and re-run `pio remote agent start`.

lkankowski, 2021-12-15:
Well, still no luck.
The first approach on RPI3B+:
```
pi@homeautomation:~ $ uname -a
Linux homeautomation 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
pi@homeautomation:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
...
pi@homeautomation:~ $ rm -rf ~/.platformio/penv
pi@homeautomation:~ $ rm -rf ~/.platformio/packages/contrib-pysite
pi@homeautomation:~ $ python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/ge                                                  t-platformio.py)"
Installer version: 1.1.0
Platform: Linux-5.10.60-v7+-armv7l-with-glibc2.28
Python version: 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/pi/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting platformio
  Downloading platformio-5.2.4.tar.gz (219 kB)
     |████████████████████████████████| 219 kB 534 kB/s
  Preparing metadata (setup.py) ... done
Collecting bottle==0.12.*
  Downloading https://www.piwheels.org/simple/bottle/bottle-0.12.19-py3-none-any.whl (89 kB)
     |████████████████████████████████| 89 kB 399 kB/s
Collecting click!=8.0.2,<9,>=8
  Downloading https://www.piwheels.org/simple/click/click-8.0.3-py3-none-any.whl (97 kB)
     |████████████████████████████████| 97 kB 491 kB/s
Collecting colorama
  Downloading https://www.piwheels.org/simple/colorama/colorama-0.4.4-py2.py3-none-any.whl (16 kB)
Collecting marshmallow<4,>=2
  Downloading https://www.piwheels.org/simple/marshmallow/marshmallow-3.14.1-py3-none-any.whl (47 kB)
     |████████████████████████████████| 47 kB 343 kB/s
Collecting pyelftools<1,>=0.27
  Downloading https://www.piwheels.org/simple/pyelftools/pyelftools-0.27-py2.py3-none-any.whl (151 kB)
     |████████████████████████████████| 151 kB 692 kB/s
Collecting pyserial==3.*
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
     |████████████████████████████████| 90 kB 549 kB/s
Collecting requests==2.*
  Downloading https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 103 kB/s
Collecting semantic_version==2.8.*
  Downloading https://www.piwheels.org/simple/semantic-version/semantic_version-2.8.5-py2.py3-none-any.whl (19 kB)
Collecting tabulate==0.8.*
  Downloading https://www.piwheels.org/simple/tabulate/tabulate-0.8.9-py3-none-any.whl (25 kB)
Collecting zeroconf==0.37.*
  Using cached https://www.piwheels.org/simple/zeroconf/zeroconf-0.37.0-py3-none-any.whl (97 kB)
Collecting aiofiles==0.8.*
  Using cached https://www.piwheels.org/simple/aiofiles/aiofiles-0.8.0-py3-none-any.whl (13 kB)
Collecting ajsonrpc==1.*
  Downloading https://www.piwheels.org/simple/ajsonrpc/ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette==0.17.*
  Downloading https://www.piwheels.org/simple/starlette/starlette-0.17.1-py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 306 kB/s
Collecting uvicorn==0.16.*
  Using cached https://www.piwheels.org/simple/uvicorn/uvicorn-0.16.0-py3-none-any.whl (56 kB)
Collecting wsproto==1.0.*
  Downloading https://www.piwheels.org/simple/wsproto/wsproto-1.0.0-py3-none-any.whl (24 kB)
Collecting idna<4,>=2.5
  Downloading https://www.piwheels.org/simple/idna/idna-3.3-py3-none-any.whl (64 kB)
     |████████████████████████████████| 64 kB 401 kB/s
Collecting certifi>=2017.4.17
  Downloading https://www.piwheels.org/simple/certifi/certifi-2021.10.8-py2.py3-none-any.whl (151 kB)
     |████████████████████████████████| 151 kB 430 kB/s
Collecting charset-normalizer~=2.0.0
[... 381 lines cut ...]
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting constantly>=15.1
  Using cached constantly-15.1.0.tar.gz (21 kB)
  Preparing metadata (setup.py) ... done
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/hyperlink/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb5090760>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/hyperlink/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb504de80>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/hyperlink/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb504def8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/hyperlink/
[... cut ...]
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb50268b0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/hyperlink/
ERROR: Could not find a version that satisfies the requirement hyperlink>=17.1.1 (from twisted) (from versions: none)
ERROR: No matching distribution found for hyperlink>=17.1.1
Error:

Please ensure that the next packages are installed:

sudo apt install python3-dev libffi-dev libssl-dev
```
Is there any documentation with complete list of required dependencies?

gdanov, 2022-02-10:
after fighting with this on two different arm distros and boards, what works for me is:

- install platform io and prerequisites as in the official docs
- `apt install rustc`
- `pip3 install cryptography==3.4.8 --no-binary cryptography` (use the highest accepted current version)

only after that I run `pio account login` and `pio remote agent start` and everything's fine.

alex, 2022-02-11:
cryptography 3.4 uses rust by default (but it can be disabled with the `CRYPTOGRAPHY_DONT_BUILD_RUST` environment variable).

Please note that if you pin to an older verison you will not receive security updates.

ivankravets, 2022-02-11:
@alex thanks for the help! It was a bug on our side, we [set `CRYPTOGRAPHY_DONT_BUILD_RUST` to the process' environment](https://github.com/platformio/platformio-core/blob/develop/platformio/package/manager/core.py#L140) but didn't pass to the subprocess. Fixed in https://github.com/platformio/platformio-core/commit/526abc6a9f21bfc0fca000c40a1f58cbff0343a1

@gdanov could you remove `rustc` (`apt purge rustc`), remove `~/.platformio/packages/contrib-pysite`, and re-start agent. Does it work now?

closed 2022-02-11 by ivankravets

ivankravets, 2022-02-11:
> Please note that if you pin to an older verison you will not receive security updates.

We can't use the latest `cryptography`, there are Linux distros for card-sized PCs that do not have `rustup`/`rustc` in the registry. Hope you will allow again to use the latest `cryptography` without dependency on Rust.

alex, 2022-02-11:
We will not be removing our rust dependency.

My recommendation to folks would be to build wheels on a larger machine and then make them available to smaller machines, e.g. https://www.piwheels.org/

lkankowski, 2022-02-12:
@gdanov thank you for workaround.
@ivankravets @alex Thank you for solving the problem.

ivankravets, 2022-02-12:
@lkankowski could you try the latest version of PlatformIO Core via `pio upgrade --dev` without any workarounds? Does it work?

lkankowski, 2022-02-12:
@ivankravets I have tried last stable version after trying workaround and then removed the rustc and it always finishes with error, but the second run is fine (it wont start to compile and everything is fine).

But for you and all your help I will try once more :)

lkankowski, 2022-02-12:
@ivankravets here is the result
https://pastebin.com/MFbYnVnd

ivankravets, 2022-02-12:
@lkankowski it works, great! The moment that the first command fails is a known issue and we will soon it later.

gdanov, 2022-02-12:
starting with clean `~/.platformio` and `pio upgrade -dev` the first run of `pio remote agent start` fails with the usual error 
```
ImportError: cannot import name '_get_backend' from 'cryptography.hazmat.backends' (/root/.platformio/packages/contrib-pysite/cryptography/hazmat/backends/__init__.py)
```
the second run of `pio remote agent start` is successful.

commit 622a190a (2021-03-15): Avoid "rustup" when building cryptography for contrib-pysite // Resolve #3865
commit a4692d54 (2021-12-14): Improved PIO Remote setup on credit-card sized computers (Raspberry Pi, BeagleBon, etc) // Resolve #3865
commit 526abc6a (2022-02-11): Improved PIO Remote setup on credit-card sized computers (Raspberry Pi, BeagleBon, etc) // Resolve #3865