platformio/platformio-core #3866: Can't Login to all PIO's services

Kamilake, 2021-03-02:
What kind of issue is this?

- [x] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10 20H2 with WSL Ubuntu 20.10 (Groovy Gorilla)

VScode 1.53.2
**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 5.1.0

### Description of problem

#### Steps to Reproduce

1. Install Platformio.
2. Log In with PlatformIO Account or Log In with OAuth
[image]

### Actual Results
Error output at the top "code not valid"

When I use GitHub login, I get a message like this:
[image]

### Expected Results
Retrieve my information with successful login.

### If problems with PlatformIO Build System:

### Additional info

I am a new user of pio. And I tried to create a new account and it failed.
I don't have permission to write on pio forums, and I can't write on Reddit.

Because they all require login.

I will provide any additional information needed.
I apologize for the lack of information.

ivankravets, 2021-03-02:
Could you try again?

Kamilake, 2021-03-03:
> Could you try again?

Thank you for replying.
It still doesn't work and when I click the login button it prints 'Not Found' at address 127.0.0.1.
[image]
[image]

ivankravets, 2021-03-03:
Could you try to remove all cookies or open that URL in the private window in Google chrome?

Kamilake, 2021-03-04:
> Could you try to remove all cookies or open that URL in the private window in Google chrome?
Thanks for the reply.
[image]

[image]

Deleting cookies didn't work for me.
But the New Incognito tab worked successfully.
Although the problem has not been completely resolved, it has made the official forum accessible. Thanks! I'm going to write there.

Kamilake, 2021-03-04:
[image]
[image]

But still I can't log in from VScode-WSL.
When you click the login button, the following window is displayed.
I used the Chrome built-in cookie, clear cache feature, and I used the EditThisCookie extension. This problem could not be solved using Incognito.

Please do not hesitate to let me know if you have any additional information you need.

ivankravets, 2021-03-10:
Have you resolved this issue?

Kamilake, 2021-03-11:
> Have you resolved this issue?

No, it got worse.
[image]
Now the browser is not showing up..

This problem only occurs in WSL.
so I have given up on WSL for now.

ivankravets, 2021-03-14:
WSL is not currently supported.

closed 2021-03-14 by ivankravets