platformio/platformio-core #3870: Error: No such command "home" when running pio -h

foopub, 2021-03-03:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Void Linux

**PlatformIO Version** (`platformio --version`):  PlatformIO Core, version 5.1.0

### Description of problem

```
$ pio -h
Usage: pio [OPTIONS] COMMAND [ARGS]...
Try 'pio -h' for help.

Error: No such command "home"
```
I've tried using `--help`, `help` as well as `platformio`.

#### Steps to Reproduce

Install PlatformIO, run `pio -h`. 
Tested in TTY using dash shell.

This seems like a glaring problem but googling "Error: No such command home" returns a single result which isn't relevant, so apologies if it's already been reported.

ivankravets, 2021-03-04:
How did you install PlatformIO Core? Please uninstall ALL PlatformIO Cores https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system

We recommend using PlatformIO IDE for VSCode or installer script https://docs.platformio.org/en/latest/core/installation.html#installer-script

closed 2021-03-04 by ivankravets

foopub, 2021-03-05:
I installed it via my distro's packet manager. I've reinstalled it since the error  As I said:

>  I've been using PIO for a long time and it worked fined before the last update.

The error appears here:
```py
> /usr/lib/python3.9/site-packages/click/core.py(1185)format_commands()
-> cmd = self.get_command(ctx, subcommand)
(Pdb) s
--Call--
> /usr/lib/python3.9/site-packages/platformio/commands/__init__.py(58)get_command()
-> def get_command(self, ctx, cmd_name):
(Pdb) r
--Return--
> /usr/lib/python3.9/site-packages/platformio/commands/__init__.py(70)get_command()->None
-> raise click.UsageError('No such command "%s"' % cmd_name, ctx)
(Pdb) p cmd_name
'home'
```
Essentially this checks every single subcommand (that I may never even use) before running what I want. I've never once used the IDE/vscode nor do I plan to, and doing so won't solve root of the problem, it will simply add "home" to the available subcommands.

ivankravets, 2021-03-05:
What is your distro? Where can we find this package with PlatformIO Core?

1. Please remove this PlatformIO Core package installed from distro
2. sudo apt-get install python3-venv
3. https://docs.platformio.org/en/latest/core/installation.html#installer-script