platformio/platformio-core #3875: serial.serialutil.SerialException: could not open port 'COM5': FileNotFoundError

WhiteLionATX, 2021-03-07:
All my projects (espressif32/espressif8266) drop this error while "upload_port = COM8" <- thats the right one.
I guess this issue was caused by an update.

`Traceback (most recent call last):
  File "C:\Users\Admin\.platformio\packages\tool-esptoolpy\esptool.py", line 3969, in <module>
    _main()
  File "C:\Users\Admin\.platformio\packages\tool-esptoolpy\esptool.py", line 3962, in _main
    main()
  File "C:\Users\Admin\.platformio\packages\tool-esptoolpy\esptool.py", line 3551, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\Admin\.platformio\packages\tool-esptoolpy\esptool.py", line 271, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\Admin\.platformio\penv\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "c:\users\Admin\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM5': FileNotFoundError(2, 'Das System kann die angegebene Datei nicht finden.', None, 2)`

ivankravets, 2021-03-09:
It looks like other process uses this COM port. Try to reboot machine and try again.

closed 2021-03-09 by ivankravets

WhiteLionATX, 2021-03-10:
> It looks like other process uses this COM port. Try to reboot machine and try again.

The are no devices connected to other COM ports than COM8. I restarted multiple times. Debug works like charm while upload doesn´t. Nothing helped. Any other ideas that could help me to find the problem ? - Thx!

WhiteLionATX, 2021-03-11:
> It looks like other process uses this COM port. Try to reboot machine and try again.

I MADE DOWN GRADE TO V2.21 AND THAT FIXED IT ! SO MARK THIS AS BUG PLEASE!!!

ivankravets, 2021-03-11:
what is 2.21?

WhiteLionATX, 2021-03-11:
PlatformIO V2.2.1

ivankravets, 2021-03-11:
This issue is not related to PlatformIO IDE for VSCode. You could potentially have opened serial monitor sessions in the VSCode terminal

SpacewormSera, 2022-06-12:
I have the same error even after restarting windows and I know for sure that no one else uses this port

SpacewormSera, 2022-06-12:
I found out what was my problem: in platformio.ini i wrote "upload_port = 9"  but after I changed it to "port = 9"  all worked!