platformio/platformio-core #3876: No such file or directory “Wifi.h and WiFiClientSecure.h” with Chain+

MohammedNoureldin, 2021-03-08:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`):

### Description of problem

I am trying to add WebSocket to my ESP32 project, but unfortunately the compiler fails as soon set `lib_ldf_mode` to `chain+`. I believe this used to work in my ESP8266 projects. Following is the error message I get:

> ...  
> Indexing .pio\build\esp32_dev\libc94\libArduinoJson.a
Compiling .pio\build\esp32_dev\lib748\FS\FS.cpp.o
Compiling .pio\build\esp32_dev\lib748\FS\vfs_api.cpp.o
In file included from .pio/libdeps/esp32_dev/WebSockets/src/WebSocketsClient.h:28:0,
                 from include/websocket/websocket_client.hpp:3,
                 from src/websocket_client.cpp:1:
.pio/libdeps/esp32_dev/WebSockets/src/WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory
> ...  
> .pio/libdeps/esp32_dev/WebSockets/src/WebSockets.h:189:18: fatal error: WiFi.h: No such file or directory  
> ...

Below is my `platformio.ini` (the important part):

```
[env:esp32_dev]
framework = arduino
platform = espressif32
board = esp32dev
platform_packages = 
	; Uses new tool chain with a non-released Arduino package to be able to compile with C++17 (C++17 may become the default in the future, therefore this may be soon not necessary).
	toolchain-xtensa@~5.100200.0
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
build_unflags =
	-std=c++11
	-fno-exceptions ; Removes the default exceptions disabling flag.
build_flags =
	-std=c++17
lib_ldf_mode = chain+
lib_deps =
        ...
        ; Here I am using the not-released version, because I need a fix for ESP32.
	https://github.com/Links2004/arduinoWebSockets.git#ed685e551f9a1a992037f5fdc791a36c4c43abc0
```

The problem seems now to occur only when adding `arduinoWebSocket` library, nevertheless this seems to be a PlatoformIO bug. Could you check that please?

### Actual Results

Project does not compile with errors shown above when setting `lib_ldf_mode` to `chain+`.

### Expected Results

Project should compile even after setting `lib_ldf_mode` to `chain+`.

### If problems with PlatformIO Build System:

**Source file to reproduce issue:**

I can reproduce the problem even without even adding and specific code, only by referencing the arduinoWebSocket library. So defining a new variable of type `WebSocketsClient` in the main.cpp file should be more than enough.

ivankravets, 2021-03-09:
Please provide a simple project to reproduce this issue.

ivankravets, 2021-03-15:
Please reopen if you still need help

MohammedNoureldin, 2021-03-15:
Hi @ivankravets, sorry for being late. I did not provide any code because any empty main (or default) file is all what you need to reproduce the problem. You just need the following platformio.ini:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    https://github.com/Links2004/arduinoWebSockets.git#a14b6b73b4f05e189ca49d6e84202c2b55db528a
lib_ldf_mode = chain+
```

ivankravets, 2021-03-15:
Did you include the main header of the library? YOu don't need to use `lib_ldf_mode = chain+`

MohammedNoureldin, 2021-03-15:
@ivankravets, why I do not need `lib_ldf_mode = chain+`? I need it for my purpose, just in this specific minimal example it is not needed.

Sure I included the main header of the library. Otherwise I could not use the library. Probably you mean `#include <WebSocketsClient.h>`.

Did I understand correctly?

ivankravets, 2021-03-15:
Please provide a project in the ZIP archive to reproduce this issue.

MohammedNoureldin, 2021-03-15:
[Test_Project.zip](https://github.com/platformio/platformio-core/files/6142479/Test_Project.zip)

Here I uploaded a very very minimal project, you just need to try to rebuild, then you will see the error.

MohammedNoureldin, 2021-03-15:
@ivankravets the developer of the library reproduced the error. As stated in the issue I opened in the library repo:

https://github.com/Links2004/arduinoWebSockets/issues/635#issuecomment-799608854

ivankravets, 2021-03-16:
We reproduced this issue, thanks! There is no ETA when we will investigate and fix it. 

@valeros will back with a temporary solution.

MohammedNoureldin, 2021-03-16:
@valeros @ivankravets Thank you! I am looking then forward to hearing from you.

valeros, 2021-03-16:
Hi @MohammedNoureldin ! For some reason LDF fails to resolve dependencies in the `WebSockets` header in the `+` modes. Apparently, the easiest way is not to use `+` at all, but if you want to keep the `chain+` mode for the rest of your project then it gets a bit tricky. At the moment I see here only one possible workaround - modify the `library.json` file for the `WebSockets` library. You can either fork the library and specify new link in your `platformio.ini` or clone it directly to your `lib` folder, but in both cases you need to manually change the LDF mode to `chain`:
```json
{
    ...
    "version": "2.3.6",
    "build": {
        "libLDFMode": "chain"
    }
   ...
}
```
Sorry for the inconvenience.

MohammedNoureldin, 2021-03-16:
Thank you @valeros! I will change the `library.json` file of the WebSocket in the time being until you find the bug. Thank you for your support!

ivankravets, 2022-04-02:
Duplicate of https://github.com/platformio/platformio-core/issues/3113

closed 2022-04-02 by ivankravets

lemonkey, 2022-06-27:
@ivankravets This is still an issue for me with the latest version of platformio (6.0.2) and the `links2004/WebSockets @ 2.3.7` library.

I need to use chain+.

lemonkey, 2022-06-27:
After my fork, I am now seeing the correct dependency tree for WebSockets

|   |-- WebSockets @ 2.3.7+sha.b15e519
|   |   |-- WiFi @ 1.0
|   |   |-- WiFiClientSecure @ 1.0
|   |   |   |-- WiFi @ 1.0

Hope this helps others...

ivankravets, 2022-06-27:
Could you try the latest PIO Core 6.0.3-dev? Please use `pio upgrade --dev`. Does it work for you?