platformio/platformio-core #3882: Dependancy issues with Preproccessor directives

Awbmilne, 2021-03-13:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version**: v5.0.4

### Description of problem

When building a project with preprocessor directives that optionally include certain libraries, Libraries that are **NOT** needed/wanted are included when compiling. This creates issues when you have two libraries that define the same symbols with different values.

```c++
#if SOMETHING
  #include <A_library.h>
#else
  #include <B_library.h>
#endif
```

#### Steps to Reproduce

This issue reared its head when compiling Networking code for a Teensy 4.1.
Compiling the below file set should recreate the issue:

`main.cpp`
```c++
#include <Arduino.h>

#ifdef __IMXRT1062__ //Processor ID for Teensy 4 and Teensy 4.1
    #include "NativeEthernet.h"
#else
    #include "Ethernet.h" // Comment this line out to see the expected result
#endif

uint8_t MAC[6] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress IP = {192,168,0,100};

void setup(){
    Serial.begin(9600);
    while(!Serial);
    Ethernet.begin(MAC, IP);

    // Hardware checks from Teensy Examples
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
        Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
        while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
        }
    }
    Serial.println("No Issues!");
}

void loop(){

}
```

`platformio.ini`
```
[env:default]
platform = teensy
board = teensy41
framework = arduino
```

### Actual Results

[image]

### Expected Results

[image]

### Additional info
While this is _generally_ not an issue, In specific cases like this one, This creates an issue where the wrong version of `Ethernet.hardwareStatus()` is used (along with the rest of the ethernet functions), meaning the Ethernet connection doesnt work. This is best demonstrated by the Serial output of the above script.

**Expected**

[image]

**Actual**

[image]

liquiddandruff, 2021-03-13:
See https://docs.platformio.org/en/latest/librarymanager/ldf.html?highlight=dependency#c-c-preprocessor-conditional-syntax

In short, try the `chain+` mode:

```
[env:myenv]
lib_ldf_mode = chain+
```

ivankravets, 2021-03-14:
Thanks, @liquiddandruff for the solution!

closed 2021-03-14 by ivankravets

Awbmilne, 2021-03-16:
Really appreciate it @liquiddandruff ! 🥂