platformio/platformio-core #3887: Library without includeDir unpredictable

dzarda, 2021-03-17:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** version 5.1.1b1

### Description of problem
Referencing [CNL](https://github.com/johnmcfarlane/cnl) (not containing library.json) causes contents of `include/cnl/` to be placed in `.pio\libdeps\penboard\cnl`. The `includeDir` is also aimed at that directory (`-I.pio\libdeps\penboard\cnl`). This breaks the intended library usage with `#include "cnl/....h"`. AFAIK this is in contrast with `includeDir` documentation that talks about *library/include* being the default. Same behaviour is seen both with library.json missing or just `includeDir` missing.

### Actual Results
The subdirectory `include/cnl` is extracted to libdeps/cnl, making the `cnl/...h` include invalid.

### Expected Results
`cnl/...h` include should be the right option.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:bla]
platform = native
lib_deps = https://github.com/johnmcfarlane/cnl/archive/v1.x.zip
```

**Source file to reproduce issue:**
```cpp
#define CHAR_BIT 8
#include "cnl/all.h"

int main()
{
}

```

### Additional info

I probably understand the logic behind the behaviour, but I think there;s still an inconsistency with the docs.

- Either fix the docs to document the behaviour with looking for directories containing `.h` files
- Or always add the `library/include` path as per docs (preferred)

ivankravets, 2021-03-17:
See https://docs.platformio.org/en/latest/librarymanager/config.html#includedir

It seems you need `"includedir": "."`

dzarda, 2021-03-17:
@ivankravets I don't think you understand. This is about the behaviour when I'm trying to use a foreign library without library.json in it

ivankravets, 2021-03-17:
@valeros could you reproduce this issue?

maxgerhardt, 2021-03-21:
Maybe partly related to https://community.platformio.org/t/platformio-downloads-subfolder/20154. In general, the behavior on how PlatformIO downloads a library is sometimes very wrong -- in the linked topic you can e.g. see that for the repo https://github.com/digint/tinyfsm, when using it in `lib_deps`, downloads only the content of the first subfolder of the examples folder (https://github.com/digint/tinyfsm/tree/master/examples/api). 

Maybe something in general can be done here to improve the logic in which subfolder of a package is actually used, in case no `library.json` or `library.properties` is present.

Regarding the original problem: I've forked the CNL library and added the `library.json` to it. You can use it per 

```
[env:native]
platform = native
lib_deps = https://github.com/maxgerhardt/cnl.git#v1.1.7-pio
build_flags = -std=c++17
```

(I've only added it to this branch since I couldn't get the 2.0.0-dev branch working, seems to need C++20)

dzarda, 2021-03-21:
I think we need a way of injecting metadata to a foreign raw dependency (Conan allows this).

Maybe there's a way with extra_scripts, but that's entirely unobvious to me.

ivankravets, 2021-09-29:
Thanks for the report! Please re-test with `pio -upgrade --dev`.

commit e8769fff (2021-09-29): Improved handling of a library root based on "Conan" or "CMake" build systems // Resolve #3887
closed 2021-09-29 by ivankravets