platformio/platformio-core #3893: Debugging using Blackmagic probe is broken

Wire7, 2021-03-18:
I was able to debug my STM32 bluepill using Blackmagic probe but suddenly I started getting errors. I am able to upload firmware to the target board (STM32 Bluepill) but when I start a debugging session I get the following error!

```
undefinedError: Traceback (most recent call last):
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/platformio/main.py”, line 109, in main
cli() # pylint: disable=no-value-for-parameter
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/click/core.py”, line 829, in call
return self.main(*args, **kwargs)
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/click/core.py”, line 782, in main
rv = self.invoke(ctx)
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/platformio/commands/init.py”, line 44, in invoke
return super(PlatformioCLI, self).invoke(ctx)
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/click/core.py”, line 1259, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/click/core.py”, line 1066, in invoke
return ctx.invoke(self.callback, **ctx.params)
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/click/core.py”, line 610, in invoke
return callback(*args, **kwargs)
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/click/decorators.py”, line 21, in new_func
return f(get_current_context(), *args, **kwargs)
File “/Users/engmoh/.platformio/penv/lib/python3.8/site-packages/platformio/commands/debug/command.py”, line 117, in cli

    server_executable = debug_options["server"]["executable"].lower()
TypeError: ‘NoneType’ object is not subscriptable

============================================================

An unexpected error occurred. Further steps:

Verify that you have the latest version of PlatformIO using
pip install -U platformio command

Try to find answer in FAQ Troubleshooting section
https://docs.platformio.org/page/faq.html

Report this problem to the developers
https://github.com/platformio/platformio-core/issues

```

My Environment:

-Mac Os big sur.
-vscode (Version: 1.54.3).
-PlatformIO (Core 5.1.1 - Home 3.3.4).
-BlackMagic probe clone (bluepill converted!) <-- Used to be working before the latest update.

This is the platformio.ini configuration:

```ini
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = cmsis
board_build.mcu = stm32f103c8t6                 ; change microcontroller (override)
board_build.f_cpu = 8000000L                    ; change MCU frequency   (override)
upload_protocol = blackmagic                    ; SWD interface
debug_tool = blackmagic
debug_port = /dev/tty.usbmodem7BBB54AD1
upload_port = /dev/tty.usbmodem7BBB54AD1 
Debugging with a STlink V2 works normally.
```

What could be the issue here?

ivankravets, 2021-03-18:
Duplicate of https://github.com/platformio/platform-ststm32/issues/505

closed 2021-03-18 by ivankravets

Wire7, 2021-03-21:
I just tried it and I still get the same error message. What could be wrong?

ivankravets, 2021-03-21:
Do you use dev-platform from the development branch? See https://github.com/platformio/platform-ststm32/issues/505#issuecomment-801781592

Wire7, 2021-03-21:
I did replace the platform configuration with the one suggested and nothing changed 

this is my new in

```ini

[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32.git       ;Due to issues
;platform = ststm32
board = bluepill_f103c8
framework = cmsis
board_build.mcu = stm32f103c8t6                 ; change microcontroller (override)
board_build.f_cpu = 8000000L                    ; change MCU frequency   (override)
upload_protocol = blackmagic                    ; SWD interface
;upload_protocol = blackmagic-jtag              ; JTAG interface
debug_tool = blackmagic
debug_port = /dev/tty.usbmodem7BBB54AD1
upload_port = /dev/tty.usbmodem7BBB54AD1       ; There are 2 Ports, UART and SWD, Check ports in terminal% ls /dev/tty.*
;build_type = debug                             ; for release version use "release"
;build_flags = -O0 -v
;build_unflags = -Os
;debug_build_unflags = 
;debug_build_flags = -O0 -v

; Environment 2:
;[env:mydebug]                  
;build_type = debug
;some_extra_options = ...

```

ivankravets, 2021-03-21:
Please run `pio update` before. Maybe you cloned `https://github.com/platformio/platform-ststm32.git` before.

Wire7, 2021-03-21:
I did, Nothing changed

ivankravets, 2021-03-21:
Also, Please open [PlatformIO Core (CLI)](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type
```
pio upgrade --dev
pio debug --interface=gdb -x .pioinit
```

Does it work from CLI?

Wire7, 2021-03-21:
Yes. It does.

ivankravets, 2021-03-21:
Try to restart VSCode. It should work as well.

Wire7, 2021-03-21:
I restarted VSCODE and now I am getting 
```
> Executing task in folder 3-Blinking_BMP: platformio debug <

*********************************************************************************************************************************
Obsolete PIO Core v5.1.1 is used (previous was 5.2.0a3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
*********************************************************************************************************************************
Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: cmsis)
---------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (12.0.0) > BluePill F103C8
HARDWARE: STM32F103C8T6 8MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - framework-cmsis-stm32f1 4.3.1 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find a linker script for the required board! An auto-generated script will be used to link firmware!
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
MethodWrapper(["checkprogsize"], [".pio/build/bluepill_f103c8/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 28 bytes from 20480 bytes)
Flash: [          ]   0.5% (used 348 bytes from 65536 bytes)
.pio/build/bluepill_f103c8/firmware.elf  :
section              size        addr
.isr_vector           268   134217728
.text                 348   134217996
.rodata                 0   134218344
.ARM.extab              0   134218344
.ARM                    0   134218344
.preinit_array          0   134218344
.init_array             4   134218344
.fini_array             4   134218348
.data                   0   536870912
.bss                   28   536870912
._user_heap_stack    1540   536870940
.ARM.attributes        41           0
.debug_line          1405           0
.debug_info          3680           0
.debug_abbrev         770           0
.debug_aranges        112           0
.debug_str           1823           0
.debug_ranges          72           0
.debug_loc            223           0
.comment              126           0
.debug_frame          124           0
Total               10568
================================================== [SUCCESS] Took 1.12 seconds ==================================================

Terminal will be reused by tasks, press any key to close it.

```

and the same error in the debug consol, weird!!!!

Wire7, 2021-03-21:
Now I tried these settings in the extensions settings 
use a portable python 3 interpreter if available = checked 
It was a development version of platformIO core = checked

Now GUI debugging is working! 
Why?

ivankravets, 2021-03-21:
Because your PlatformIO installation is broken. You installed ton of other PlatformIO Cores. 

This will help you https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system