platformio/platformio-core #3894: Issues with latest release (a2) and monitor for ESP32

cpainchaud, 2021-03-19:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.2.0a2
Clion

### Description of problem
when trying to run a 'monitor' session after building an ESP32 project I am getting python errors

#### Steps to Reproduce

1. run   pio -c clion device monitor -e genericESP32

### Actual Results

pio -c clion device monitor -e genericESP32
Error: Traceback (most recent call last):
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\__main__.py", line 115, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\commands\device\command.py", line 196, in device_monitor
    device_helpers.register_platform_filters(platform, options=kwargs)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\commands\device\helpers.py", line 106, in register_platform_filters
    load_monitor_filter(path, options)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\commands\device\helpers.py", line 82, in load_monitor_filter
    module = load_python_module("platformio.commands.device.filters.%s" % name, path)
  File "c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\compat.py", line 64, in load_python_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\christophe.painchaud\.platformio\platforms\espressif32\monitor\filter_exception_decoder.py", line 20, in <module>
    from platformio.compat import path_to_unicode, WINDOWS, PY2
ImportError: cannot import name 'path_to_unicode' from 'platformio.compat' (c:\users\christophe.painchaud\appdata\local\programs\python\python38\lib\site-packages\platformio\compat.py)

### Expected Results

Just get it to monitor :D

### Additional info

skelstar, 2021-03-20:
I'm getting the issue. Running the serial monitor by clicking the monitor icon in the icon bar at the bottom of the vscode IDE.

`platformio --version`
`PlatformIO Core, version 5.2.0a2`

Tried uninstalling PlatformIO, removing `.platformio` folder from user_dir. No luck.

```Error: Traceback (most recent call last):
  File "c:\users\skels\.platformio\penv\lib\site-packages\platformio\__main__.py", line 115, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\skels\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\skels\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\skels\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\skels\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\skels\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\skels\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\skels\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\skels\.platformio\penv\lib\site-packages\platformio\commands\device\command.py", line 196, in device_monitor
    device_helpers.register_platform_filters(platform, options=kwargs)
  File "c:\users\skels\.platformio\penv\lib\site-packages\platformio\commands\device\helpers.py", line 106, in register_platform_filters
    load_monitor_filter(path, options)
  File "c:\users\skels\.platformio\penv\lib\site-packages\platformio\commands\device\helpers.py", line 82, in load_monitor_filter
    module = load_python_module("platformio.commands.device.filters.%s" % name, path)
  File "c:\users\skels\.platformio\penv\lib\site-packages\platformio\compat.py", line 64, in load_python_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\skels\.platformio\platforms\espressif32\monitor\filter_exception_decoder.py", line 20, in <module>
    from platformio.compat import path_to_unicode, WINDOWS, PY2
ImportError: cannot import name 'path_to_unicode' from 'platformio.compat' (c:\users\skels\.platformio\penv\lib\site-packages\platformio\compat.py)```

skelstar, 2021-03-20:
@cpainchaud I found a way around by downgrading Core to v5.1.1
I ran: `pip install -U "platformio==5.1.1"`

[This link](https://community.platformio.org/t/downgrade-core-from-4-0-to-3-x/8225) helped me work it out.

ivankravets, 2021-03-20:
Thanks for the report! Please upgrade to the latest dev using `pio upgrade --dev`.

commit 990071af (2021-03-20): Fix issue with missed compat.path_to_unicode // Resolve #3894
closed 2021-03-20 by ivankravets