platformio/platformio-core #3895: Error while trying library updates

andrebiedermann63, 2021-03-21:
### Configuration

**Operating system**: Windows 10 x64 (latest)

**PlatformIO Version** (`platformio --version`): 2.3.0
Python 3.9.2 x64

VSC:
Version: 1.54.3 (user setup)
Commit: 2b9aebd5354a3629c3aba0a5f5df49f43d6689f8
Date: 2021-03-15T10:55:45.459Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042

### Description of problem
Error Message after clicking "Libraries" and "Updates"

#### Steps to Reproduce

1. Within PlatformIO click "Libraries"
2. Click "Updates and get the Errormessage below

### Actual Results
Get this Error message:

Libraries: Updates
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n cli() # pylint: disable=no-value-for-parameter\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 262, in lib_update\r\n outdated = lm.outdated(pkg, spec)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 46, in outdated\r\n latest = self._fetch_vcs_latest_version(pkg)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 69, in _fetch_vcs_latest_version\r\n vcs = VCSClientFactory.new(pkg.path, pkg.metadata.spec.url, silent=True)\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\vcsclient.py\", line 55, in new\r\n src_dir, remote_url, tag, silent\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\vcsclient.py\", line 137, in __init__\r\n self.configure()\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\vcsclient.py\", line 146, in configure\r\n result = proc.exec_command([cls.command, \"--exec-path\"])\r\n File \"c:\\users\\work-user\\.platformio\\penv\\lib\\site-packages\\platformio\\proc.py\", line 117, in exec_command\r\n p = subprocess.Popen(*args, **kwargs)\r\n File \"C:\\Users\\WORK-USER\\.platformio\\python3\\lib\\subprocess.py\", line 800, in __init__\r\n restore_signals, start_new_session)\r\n File \"C:\\Users\\WORK-USER\\.platformio\\python3\\lib\\subprocess.py\", line 1207, in _execute_child\r\n startupinfo)\r\nFileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

### Expected Results
Showing wich libraries can be updated

Thanhks in Advance

maxgerhardt, 2021-03-21:
>c:\users\work-user\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure\r\n result = proc.exec_command([cls.command, "--exec-path"])\r\n File "c:\users\work-.

I've seen this type of traceback in the vsclient (version control client?) when it tries to execute a `git` command but git is not installed. Do you have Git installed per https://git-scm.com/downloads?

andrebiedermann63, 2021-03-22:
Hello

Git was installed. But ther seemed to be a mess with "Visual Studio Code" and "PlatformIO" somehow. I uninstalled both, removed all platformio folders and reinstalled both.
No more nags now.  👍 

Thanks for the help.

closed 2021-03-22 by andrebiedermann63