platformio/platformio-core #3896: Advanced Scripting and reading from stdin creates issues with platformio-environment

baender, 2021-03-21:
- [x] **PlatformIO IDE**.
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 Home, 20H2, 64 bit

**PlatformIO Version**: PlatformIO Core, version 5.1.1

**Python**: 3.9.2, 64 bit

**VSCode**: 1.54.3

**NodeJS**: 12.18.3

### Description of problem
 
When using Advanced Scripting with Python's `input()` for reading from stdin `PlatformIO: Rebuilding IntelliSense Index...` runs forever, not allowing me to use any tasks of the environement. `Project tasks` in the activity bar is empty and building from the status bar produces `command 'platformio-ide.build' not found`.

A work around is to actively choose an environment from the status bar, which allows to use Default project tasks.

However, when reloading the project tasks via the button from the activity bar, it gets stuck with `PlatformIO: Loading tasks...` blocking also the default environment. Nothing will do until re-opening the folder.

#### Steps to Reproduce

1. Add the following line to your environment in `platformio.ini`
`extra_scripts = pre:getName.py`

2. Create file `getName.py` with the following content
```py
Import("env")

stationName = "dummy"
# stationName = input("***Enter station name***\n")

env.Append(CPPDEFINES=[
  ("STATION_NAME", "\\\"" + stationName + "\\\"")
])

```

3. Build with `stationName = "dummy"`
4. Switch to `stationName = input("***Enter station name***\n")` and build

### Actual Results

When using `dummy`: everything works as expected.
When using `input()`: `PlatformIO: Rebuilding IntelliSense Index...` runs forever and no tasks for the environment are available.

### Expected Results

It should behave the same in both cases.

**The content of `platformio.ini`:**
```ini
[env:mkrnb1500]
platform = atmelsam
board = mkrnb1500
framework = arduino
extra_scripts =
  pre:getName.py
```

**Source file to reproduce issue:** (does not matter)
```cpp
#include <Arduino.h>

void setup() {
    Serial.begin(9600);
    while(!Serial);
}

void loop() {
    Serial.print("This station is called: ");
    Serial.println(STATION_NAME);
    delay(5000);
}

```

### Additional info

On a side note:

The [documentation to Advanced Scripting](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html) has the example:

```py
env.Append(CPPDEFINES=[
  "MACRO_1_NAME",
  ("MACRO_2_NAME", "MACRO_2_VALUE")
])
```

This does not compile as MACRO_2_VALUE is used as an identifier in the .cpp not as a string. To make it work I had to inject the escape character and the quotes resulting in:

```py
env.Append(CPPDEFINES=[
  "MACRO_1_NAME",
  ("MACRO_2_NAME", "\\\"MACRO_2_VALUE\\\"")
])
```

If that is the desired behaviour, perhaps one wants to add this piece of information to the documentation of advanced scripting, as it is misleading the way it is written at the moment.

PS.:
Thank you for your wonderful piece of software!

ivankravets, 2021-03-22:
Sorry, I can't reproduce this issue

[image]

ivankravets, 2021-03-22:
> Thank you for your wonderful piece of software!

Happy coding with PlatformIO!

closed 2021-03-22 by ivankravets