platformio/platformio-core #3899: PIO finds "GFX Library for Arduino" library, breaks Adafruit GFX firmwares

maxgerhardt, 2021-03-23:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.2.0a3`

### Description of problem

For some reason PlatformIO picks up this library: https://www.arduinolibraries.info/libraries/gfx-library-for-arduino (which isn't even listed in https://platformio.org/lib/search?query=GFX%20Library%20for%20Arduino) when compiling a standard test sketch for one of the Adafruit GFX / TFT libraries.

#### Steps to Reproduce

1. Per https://community.platformio.org/t/adafruit-gfx-and-esp8266/20207/3?u=maxgerhardt

### Actual Results

>.pio\libdeps\d1_mini_pro\GFX Library for Arduino\src\Arduino_TFT.cpp: In member function 'virtual void Arduino_TFT::drawChar(int16_t, int16_t, unsigned char, uint16_t, uint16_t)':
>.pio\libdeps\d1_mini_pro\GFX Library for Arduino\src\Arduino_TFT.cpp:1125:1: error: a15 cannot be used in asm here
 >}
 >^

And lib dep graph

```
Dependency Graph
|-- <Adafruit ST7735 and ST7789 Library> 1.6.1
|   |-- <Adafruit GFX Library> 1.10.6
|   |   |-- <SPI> 1.0
|   |   |-- <Adafruit BusIO> 1.7.2
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <GFX Library for Arduino> 1.0.8
|   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.7.2
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit seesaw Library> 1.3.1
|   |-- <Wire> 1.0
|-- <Adafruit GFX Library> 1.10.6
|   |-- <SPI> 1.0
|   |-- <Adafruit BusIO> 1.7.2
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
```

I think it's being picked up due to exposing the `gfxfont.h` header file, which is also in `Adafruit GFX Library`. 

I'm not sure what's the best way to handle this (library priority? Check if a header file *is* already satisfied by one of the dependencies, and then don't search for a dependency filling in that header?), but it's definitely breaking compilations right now.

### Expected Results

Normal compilation.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
monitor_speed = 115200
lib_deps = 
	adafruit/Adafruit ST7735 and ST7789 Library@^1.6.1
	moononournation/GFX Library for Arduino@^1.0.7
	adafruit/Adafruit BusIO@^1.7.2
	adafruit/Adafruit seesaw Library@^1.3.1
```

**Source file to reproduce issue:**
As a **ino** file: https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/examples/graphicstest/graphicstest.ino

### Additional info

Workaround is to `lib_ignore = GFX Library for Arduino` but people have to find this workaround first after having their compilation broken with the release and pickup of that library.

valeros, 2021-09-28:
@maxgerhardt Strange, but I cannot reproduce the issue, I put together a simple project with the `platformio.ini` as described above and graphicstest.ino` in the `src` folder and it compiled just fine (only needed to add `#include <Wire.h>`). Could you please try to run the same project but using the latest PlatformIO core and Espressif8266 dev-platform v3.1? Thanks in advance.

maxgerhardt, 2021-09-28:
Indeed not reproducable anymore. Closing.

closed 2021-09-28 by maxgerhardt