platformio/platformio-core #3900: Library generation - TIme and SD handlers

brunolnetto, 2021-03-24:
### Configuration

Linux
PlatformIO Core 5.1.1

### Description of problem

Hi,

I try to upload two libraries without success. They call "Einstein" and "FortuneTeller". The former provides a user-friendly time string from millis() command. The latter formats and manages SD cards content. I setup both with recommended config files. However, both fail to reproduce minimal examples I created. I utilize a Arduino UNO board, easy to use at home. The github URL are respectively below:

https://github.com/brunolnetto/einstein
https://github.com/brunolnetto/fortuneteller

#### Steps to Reproduce

1. Download VS Code;
2. Install PlatformIO
3. Go to Libraries section and install both libraries
4. Build and run the examples

### Actual Results

/tmp/ccLUQqK2.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x86): undefined reference to `setup'
<artificial>:(.text.startup+0x8e): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/uno/firmware.elf] Error 1

### Expected Results

No failed result.
Serial monitor with time string printed.

### If problems with PlatformIO Build System:

[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps = painlessmesh/arduinoUnity@^0.0.2

**Source file to reproduce issue:**
```
#include <Arduino.h>
#include <converters/fundamental.h>

long int TIME_SPAN = 1000;
long int BAUD_RATE = 9600;

void setup(){
	Serial.begin(BAUD_RATE);
}

void loop() {
	Serial.println(now());
	delay(TIME_SPAN);
}
```

valeros, 2021-03-25:
Hi @brunolnetto ! The error message sounds pretty descriptive, so I'm not sure I understand your problem. Both links https://github.com/brunolnetto/einstein and https://github.com/brunolnetto/fortuneteller seem like broken PlatformIO projects. If they are libraries, then they cannot be compiled as standalone projects. But if that links represent projects where are the `platformio.ini` file and the main source file that contains `setup` and `loop`.

valeros, 2021-03-25:
This repo is not the right place to ask such questions, please forward it to https://community.platformio.org/

closed 2021-03-25 by valeros