platformio/platformio-core #3901: "pio test" does not install platform, leading to "Error: Unknown development platform"

maxgerhardt, 2021-03-25:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.2.0a3`

### Description of problem

Attempting to execute a `pio test` command for a `platform` that is not yet locally installed gives

```
$ pio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
------------------------------------------------------------------------------------------------------------------------
Error: Unknown development platform 'native'
```

If one does `pio platform install native` and then reruns the command, it works.

#### Steps to Reproduce

1. Create an arbitrary new project (e.g. `pio init -b uno`) 
2. Overwrite the `platformio.ini` with 
```
[env:native]
platform = native
```
3. Create a file `test\test1.cpp` with content
```
#include <unity.h>

void TrivialTestCase(void)
{
    TEST_ASSERT_EQUAL(32, 32);
}

int main() {
    UNITY_BEGIN();
    RUN_TEST(TrivialTestCase);
    UNITY_END();
    return 0;
}
```
4. Delete the folder `C:\Users\<user>\.platformio\platforms\native`
5. Run `pio test -e native`

### Actual Results

```
>pio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
------------------------------------------------------------------------------------------------------------------------
Error: Unknown development platform 'native'
```

### Expected Results

Expected results are present after

```
>pio platform install native
Platform Manager: Installing native
Platform Manager: native @ 1.1.3 has been installed!
The platform 'native' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.

>pio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
test\test1.cpp:10:TrivialTestCase       [PASSED]

-----------------------
1 Tests 0 Failures 0 Ignored
OK
============================================== [PASSED] Took 0.99 seconds ==============================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       native         PASSED    00:00:00.987
============================================= 1 succeeded in 00:00:00.987 =============================================
```
### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
Per above
**Source file to reproduce issue:**
per above

### Additional info

mcspr, 2021-03-25:
will also resolve https://github.com/platformio/platform-native/issues/12 (ref. my comment https://github.com/platformio/platform-native/issues/12#issuecomment-807070747)
(*was just about to write an issue...*)

donnm, 2021-03-25:
I'm having this issue too:

```
$ pio platform install atmelavr 
PlatformManager: Installing atmelavr
Error: Detected unknown package 'atmelavr'
```

The reason this is failing is because the HTTP request to `http://dl.platformio.org/platforms/manifest.json` results in `HTTP/1.1 404 Not Found`.

For what it's worth the error message `Error: Detected unknown package 'atmelavr'` could be more descriptive.

maxgerhardt, 2021-03-25:
What is `pio --version`? Some 4.x version or 5.x?

donnm, 2021-03-25:
PlatformIO, version 4.3.4 for me.

maxgerhardt, 2021-03-25:
Mhm I have a feeling some of the stuff for the old 1-year old 4.x core was shut down :O. Is it viable for you to try and upgrade thet PlatformIO core to 5.x? (`pio upgrade`)

donnm, 2021-03-25:
@maxgerhardt Upgrading to 5.1.1 fixed my issue. +1 for backwards compatibility.

donnm, 2021-03-25:
I can confirm this issue with 5.1.1 on Debian:

```bash
$ pio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
---------------------------------------------------------------------------------------------
Error: Unknown development platform 'native'
$ pio --version
PlatformIO Core, version 5.1.1
$ cat platformio.ini 
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino

[env:native]
platform = native
$ ls ~/.platformio/platforms/
atmelavr  espressif8266
```

maxgerhardt, 2021-03-25:
And it is confirmed fixed after you've run `pio platform install native` right?

donnm, 2021-03-25:
Yes.

ivankravets, 2021-04-01:
Sorry for the issue. Please re-test with `pio upgrade --dev`.

@puzrin , we would be thankful if you switch your CI to PlatformIO Core Dev. We have a lot of tests and each our commit is very stable but sometime issues happen. So, could add this line 
```
pio upgrade --dev
```

after `pip install -U platformio`? in https://github.com/speedcontrols/ac_sc_grinder/blob/master/.github/workflows/ci.yml#L19 The final code:

```
          pip install -U platformio
          platformio upgrade --dev          
          platformio update
```

closed 2021-04-01 by ivankravets

puzrin, 2021-04-01:
Tested in branch. It works https://github.com/speedcontrols/ac_sc_grinder/actions/runs/709580960.

ivankravets, 2021-04-01:
@puzrin please keep `--dev` branch and report us if you encounter any issues. Thanks!

commit ee7ea77f (2021-04-01): Fixed an error "Unknown development platform" when running unit tests on a clean machine // Resolve #3901