platformio/platformio-core #3903: [BUG] pio ci doesn't upload firmware

fabiuz7, 2021-03-26:
Hi,
I had found the following problem: 

**Operating system**: windows 10 (VScode)
**PlatformIO Version** (`platformio --version`): 5.1.1

### Description of problem

When I run 

    pio ci --project-conf platformio.ini --project-option="targets=upload" --project-option="upload_port=COM3" --lib src examples/esp_logger_rotate

the sketch is properly compiled, but it doesn't upload the firmware. The esp8266 works perfectly, I'm able to upload the sketch with other tools.

platformio.ini is minimal:

    [env:d1_mini]
    platform = espressif8266
    board = d1_mini
    framework = arduino
    lib_ldf_mode = deep

### Actual Results
These are the last lines printed on console:

```
Archiving .pio\build\d1_mini\libFrameworkArduino.a
Linking .pio\build\d1_mini\firmware.elf
Building .pio\build\d1_mini\firmware.bin
Retrieving maximum program size .pio\build\d1_mini\firmware.elf
Checking size .pio\build\d1_mini\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  35.3% (used 28952 bytes from 81920 bytes)
Flash: [===       ]  28.7% (used 299380 bytes from 1044464 bytes)
Creating BIN file ".pio\build\d1_mini\firmware.bin" using "C:\Users\myuser\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\d1_mini\firmware.elf"

```

Hoping I have posted all the relevant information!

fabiuz7, 2021-03-27:
I was trying to upload the firmware with esptool, but I found out that I can't see any firmware. I had added --keep-build-directory, but the entire "build" directory do not exist. The command is:

    pio ci --project-conf platformio.ini --keep-build-dir  --lib src examples/esp_logger_rotate

and the terminal emit the usual log:
    
     ....
    Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_info.c.o
    Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_integrity.c.o
    Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_local.c.o
    Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_malloc.cpp.o
    Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_poison.c.o
    ....

But build directory is not even created. --verbose option does not add any useful (to me) details. Maybe is it a problem with relative / absolute path?

fabiuz7, 2021-03-27:
Yet another update!

if I explicitly specify the build directory 

    --build-dir my-build --keep-build-directory

the firmware is created and preserved. The only "issue" is that I have to manually remove the build folder before relaunching the building process.

Still, the upload is not performed...

ivankravets, 2021-03-27:
`pio ci` is not intended for device programming. You need https://docs.platformio.org/en/latest/core/userguide/cmd_run.html

fabiuz7, 2021-03-27:
Sorry @ivankravets, but the documentation of "ci" says that you can upload the firmware.
I'm referring to

https://docs.platformio.org/en/v5.1.1/core/userguide/cmd_ci.html

commit 7e995696 (2021-04-02): Remove a note with using `pio ci` for uploading // Resolve #3903
closed 2021-04-02 by ivankravets