platformio/platformio-core #3906: Using dot `.` in PROGNAME causes unexpected behavior

MohammedNoureldin, 2021-03-27:
I want to rename the output `firmware.bin` file according to the example supplied by the official documentation.

platformio.ini:
```
[env:env_custom_prog_name]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -D VERSION=13
extra_scripts = pre:extra_script.py
```

extra_script.py:

```
Import("env")

my_flags = env.ParseFlags(env['BUILD_FLAGS'])
defines = {k: v for (k, v) in my_flags.get("CPPDEFINES")}
# print(defines)

env.Replace(PROGNAME="firmware_%s" % defines.get("VERSION"))
```

The default empty `main.cpp` file is sufficient to reproduce the problem.

The previous default example will work. However, as soon as I use the following `PROGNAME` (which include dots), the program does not build properly anymore. It removes the extension of the `,elf` file, and the `partitions.bin` is missing (which causes fail to upload).

So in other words, when I use the following `VERSION` instead of `13`: `1.2.3+4`.

The problem happens exactly because there is a dot inside the the `PROGNAME`. Could you please check it?

MohammedNoureldin, 2021-03-28:
@ivankravets should not this be labeled as bug?

MohammedNoureldin, 2021-04-01:
@ivankravets Could you or anyone give an update or configuration about this please?

ivankravets, 2021-04-02:
Try to not use dots and replace them with `_`. It seems that issue goes down to the [SCons tool](https://scons.org/).

ivankravets, 2021-04-02:
Please re-test with `pio upgrade --dev`.

closed 2021-04-02 by ivankravets

MohammedNoureldin, 2021-04-02:
@ivankravets @valeros This seems to be working now! Thanks a lot to you both!

PR #3918 (2021-04-02): Explicitly specify PROGSUFFIX when compiling final binary
Resolves #3906
commit 551bd3db (2021-04-02): Explicitly specify PROGSUFFIX when compiling final binary (#3918)

Resolves #3906