platformio/platformio-core #3907: Failure building Adafruit Sensor Calibration Library - can’t find Adafruit_SPIFlash.h

gzavagli, 2021-03-28:
### Configuration

**Operating system**: macOS Big Sur

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.1

### Description of problem
A project using Adafruit's Sensor Calibration library fails to compile, claiming that this library can't find Adafruit_SPIFlash.h.

#### Steps to Reproduce
Repro steps are available at https://community.platformio.org/t/failure-building-adafruit-sensor-calibration-library-cant-find-adafruit-spiflash-h/20294/2

valeros, 2021-03-29:
Hi @gzavagli ! IMO, it doesn't seem like PlatformIO issue, simply missing a macro which wraps `#include <Adafruit_SPIFlash.h>`. I'm wondering how it works in Arduino IDE? Have you tried to compile the same example there?

gzavagli, 2021-03-29:
I just created a small sketch in the Arduino IDE with the minimal app described at https://community.platformio.org/t/failure-building-adafruit-sensor-calibration-library-cant-find-adafruit-spiflash-h/20294/2?u=gzavagli and WAS NOT able to repro the compile issue.

I don't have a strong opinion in this matter, it's just that @maxgerhardt seemed to think this was a genuine bug in PIO.

maxgerhardt, 2021-03-30:
I'll double check this shortly, but it looked to me as if it wasn't evaluating a macro expression correctly and didn't see it was included. Full details coming soon.

ivankravets, 2022-04-02:
The issue will be solved in https://github.com/platformio/platformio-core/issues/3113

closed 2022-04-02 by ivankravets