platformio/platformio-core #3913: Old `lib_deps` downloaded are not removed when `lib_deps` is changed (for example library version update)

GunZi200, 2021-03-31:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): `5.1.1` and also `5.2.0a3`

### Description of problem

Platform IO does not remove old revisions of a library or previously downloaded library that are no longer included with `lib_deps`

I seem to recall that Platform IO should take care of this for the user no?

[image]

#### Steps to Reproduce

1. Write in `.ini `file:
```ini
[env:micro]
platform = atmelavr
board = megaatmega2560
framework = arduino

; change microcontroller
board_build.mcu = atmega2560

; change MCU frequency
board_build.f_cpu = 16000000L

; baude rate for serial monitor
monitor_speed = 115200

lib_deps =
     MQTT@2.4.7 ; MQTT library
```
2. Hit the 'Build' button in Visual Studio Code
3. Now change  `MQTT@2.4.7` to  `MQTT@2.4.8`
4. Build again
5. Open the .pio/libdeps/micro folder and see actual result:

### Actual Results

[image]

### Expected Results

Version 2.4.7 should be removed if 2.4.8 is downloaded.
[image]

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:micro]
platform = atmelavr
board = megaatmega2560
framework = arduino

; change microcontroller
board_build.mcu = atmega2560

; change MCU frequency
board_build.f_cpu = 16000000L

; baude rate for serial monitor
monitor_speed = 115200

; Disable compiler warning for unused functions
build_unflags = -Wall

build_flags = 
     -Wunused-parameter ; Turn on compiler warning when function have unused parameters
     -Wunused-variable ;  Turn on compiler warning when variable is not used

; Libraries
lib_deps =
     https://github.com/arduino-libraries/Ethernet ; Ethernet library v2.0.0
     MQTT@2.4.8 ; MQTT library
     RTClib@^1.12.5 ; Real-time clock library
     greiman/SdFat@^2.0.6 ; SD card library
     https://github.com/DFRobot/DFRobot_SHT3x ; Library for humidity and temperature sensor

lib_ignore = TinyWireM

```

**Source file to reproduce issue:**
I don't think any code is needed. Just open a blank `main.c` or `main.cpp` or `main.ino`

GunZi200, 2021-04-02:
This issue is probably low priority since PIO uses the version listed by `lib_deps`. Workaround is to delete manually the older folder. Would be nice if PIO automated this 😃 

My only annoyance with this is that the intellisense from C++ may use the older folders/versions... which made me notice this issue.

ivankravets, 2021-04-10:
Duplicate of https://github.com/platformio/platformio-core/issues/3076

closed 2021-04-10 by ivankravets