platformio/platformio-core #3917: Cannot use assembly files in MCS51 platform due to core issue

maxgerhardt, 2021-04-02:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.2.0a3`

### Description of problem

Per https://community.platformio.org/t/mcs51-assembler-programming/20410. 

When attempting to write an assembler program for a MCS51 device on Windows by creating a `src/blink.s` file, the following error occurrs

```
Building in release mode
sdcc -DF_CPU=11059200 -DHEAP_SIZE=64 -DPLATFORMIO=50200 -Iinclude -Isrc -c -o .pio\build\stc15f204ea\src\blinky.rel src\blinky.s
at 1: error 119: don't know what to do with file 'src\blinky.s'. file extension unsupported
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.8.4 #10766 (MINGW64)
```

Suddenly it wants to use `sdcc`, the C compiler, for the `.s` file, but in the special case of SDCC, it does not now how to handle assembly files.

The platform code

https://github.com/platformio/platform-intel_mcs51/blob/c52ea3a706e540df65bdc6a0caf64f1eefeb8ac9/builder/main.py#L56-L60

sets the correct `AS="sdas8051"` directive, however the code in

https://github.com/platformio/platformio-core/blob/73d4f10f4bf88dde79148583fbfcf6914810f083/platformio/builder/tools/platformio.py#L115-L117

overwrites the assembler on non-case-sensitive operating systems to SDCC, thus causing the issue.

#### Steps to Reproduce

1. Create a new project for the stc15f204ea board
2. Add a `blink.s` file with empty content
3. Hit "Build"

### Actual Results

>sdcc -DF_CPU=11059200 -DHEAP_SIZE=64 -DPLATFORMIO=50200 -Iinclude -Isrc -c -o .pio\build\stc15f204ea\src\blinky.rel src\blinky.s
>at 1: error 119: don't know what to do with file 'src\blinky.s'. file extension unsupported

### Expected Results

It uses `sdas8051` correctly 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:stc15f204ea]
platform = intel_mcs51
board = stc15f204ea
```

**Source file to reproduce issue:**
```cpp
```

### Additional info

maxgerhardt, 2021-04-02:
Addendum: This issue also applies to platform-ststm8 if someone tries to actually use the `sdasstm8` assembler with assembly files.

https://github.com/platformio/platform-ststm8/blob/712f391f99a6b767fe7f1183fdb2b7d4d239b549/builder/main.py#L26-L29

closed 2022-04-10 by ivankravets

ivankravets, 2022-04-10:
Thanks for the report! It is a very interesting issue, mainly related to the SCons. See https://github.com/SCons/scons/blob/master/SCons/Tool/asm.py#L44
As you can see, a default ASM tool declares ".s" and ".S" extensions on Windows (non-case-sensitive) for ASCOM. This is classic behavior for native compilers but not for GCC. The https://github.com/platformio/platformio-core/commit/f63b2f79e0af3c9342401df2b19468bce5bfa83e fixes this situation.

Nevertheless, we found a historical bug in ALL dev-platforms where we overwrite `ASFLAGS` with the preprocessor's data instead of `ASPPFLAGS`. We will fix it soon.

ivankravets, 2022-05-18:
Yes, it was a historical bug in SCons on Windows, and we decided to fix it on our side. See http://draskovblog.blogspot.com/2008/12/gcc-differences-between-s-and-s.html

positron96, 2022-05-18:
Ok, thanks for the link. 
Am I right that .S files are passed to `gcc` (with -x) option, and .s files are passed to `as` (also with -x option) and the latter complains about -x option? Then there is a problem for those who have rightful .s files (to be passed directly to `as`).

Also, the compilation database shows `arm-none-eabi-as.exe` for both .s and .S, despite calling different programs in reality.  Is this a bug?

ivankravets, 2022-05-28:
@positron96 thanks, fixed in https://github.com/platformio/platform-ststm32

Could you try the latest version of ST STM32 dev-platform? 

```ini
[env:development]
platform = https://github.com/platformio/platform-ststm32.git
```

Please navigate to the project folder and type `pio pkg update`.

Does it work now?

positron96, 2022-05-29:
Yes, the assembler doesn't complain about unknown arguments anymore. It is run with these arguments now:
```
arm-none-eabi-as -o .pio\build\board\src\cubemx\startup_stm32f407xx.o src\cubemx\startup_stm32f407xx.s
```

Thank you for making very useful software!

BTW, for some reason after I changed `platform` line in `platformio.ini`, `pio pkg update` did not update anything even after I removed ststm32 folder. Removing the platform folder and rebuilding did the trick.

ivankravets, 2022-05-29:
You can only update installed packages. If they have not been installed yet, you will not see any updates. 

Thanks for the kind words! 🙏🚀

commit f7c861d0 (2022-04-10): Fixed an issue when GCC preprocessor was applied to the ".s" assembly files on case-sensitive OS such as Window OS // Resolve #3917
commit f63b2f79 (2022-04-10): Fixed an issue when GCC preprocessor was applied to the ".s" assembly files on case-sensitive OS such as Window OS // Resolve #3917