platformio/platformio-core #3922: "pio debug -c" does not find specified ini file

jcw, 2021-04-08:
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

I'm not able to use the "-c" parameter to specify a ini file:

```
$ pio debug
Error: Not a PlatformIO project. `platformio.ini` file has not been found in current working directory (/Users/jcw/code/jee/monty/examples/f7/platformio.ini). To initialize new project please use `platformio project init` command
$ pio debug -c ../../platformio.ini
Error: Not a PlatformIO project. `platformio.ini` file has not been found in current working directory (/Users/jcw/code/jee/monty/examples/f7/platformio.ini). To initialize new project please use `platformio project init` command
$ ln -s ../../platformio.ini
$ pio debug -c ../../platformio.ini
Processing disco-f750 (board: disco_f750n8; platform: ststm32; framework: cmsis)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f750n8.html
PLATFORM: ST STM32 (12.1.1) > STM32F7508-DK
HARDWARE: STM32F750N8H6 200MHz, 340KB RAM, 64KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:
 - framework-cmsis 2.50501.200527 (5.5.1)
 - framework-cmsis-stm32f7 1.2.5
 - tool-ldscripts-ststm32 0.1.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
[...]
```

Am running PIO 5.1.1, MacOS 11.2.3, Intel i7.

ivankravets, 2021-04-10:
You need to pass a custom working directory, not only config. See https://docs.platformio.org/en/latest/core/userguide/cmd_debug.html#cmdoption-pio-debug-d

Does it work? If no, could you try with the latest dev-version of PIO Core via `pio upgrade --dev`. See changes to our unified debugger https://github.com/platformio/platformio-core/blob/develop/HISTORY.rst

jcw, 2021-04-10:
I'm not sure I understand - `pio run -c ../../platformio.ini` works, the working directory is `.`, as intended.
But with `pio debug ...`, I get the above error. Also tried `pio debug -d . -c ../../platformio.ini`, but get the same error.

This passes, but gives me compile errors: `pio debug -d ../.. -c ../../platformio.ini`. I think the following is happening:

* in `../../platformio.ini`, there's a `extra_configs = monty-pio.ini` line
* that `monty-pio.ini` file is in the _current_ directory, not in `../../`
* when I do `pio run -c ../../platformio.ini`, all is well, it finds that extra file
* when I do `pio debug -d ../.. <etc>`, it will start to compile, but `monty-pio.ini` will not have been found
* ... and so with `pio debug` I get compile errors, and with `pio run` everything works as intended

My expectation was that `pio debug` (and `piodebuggdb`) would work in the same way as `pio run`.

ivankravets, 2021-04-11:
Thanks for the report. The bug has been fixed. Could you re-test with `pio upgrade --dev`?

closed 2021-04-11 by ivankravets

jcw, 2021-04-11:
Bingo, it works now. Many thanks!

commit b35c5a22 (2021-04-11): Fix a broken support for custom configuration file for `pio debug` command // Resolve #3922