platformio/platformio-core #3924: Feature request: Custom Device Monitor Filters for projects not platforms

damiendr, 2021-04-09:
I find myself wanting to write a custom device monitor filter that colors the output according to the syntax of the serial protocol I use for a particular project.

Unfortunately it seems one can only write custom filters as part of a platform.

It would be nice to be able to add a "monitor" folder in a project, or directly pass the path to a Python file to the --filter option.

Esjayar, 2021-06-30:
+1 for this. I wrote a filter for debugging some communications from a device I was interfacing with and alas and alack putting it in the `monitor` folder of my project did nothing.  

So I went ahead and tweaked the device monitor loading code a bit to also load filters from projects. Probably would only take someone who knows what they are doing a few minutes, but took me 30 or so....

Here is what I changed in case anyone else wants to have this for their projects while we wait for this to (hopefully) be added.
**In commands/device/command.py:**
Replace this:
```
    if "platform" in project_options:
        with fs.cd(kwargs["project_dir"]):
            platform = PlatformFactory.new(project_options["platform"])
            device_helpers.register_platform_filters(platform, options=kwargs)
```

With this:
```
    if "platform" in project_options:
        with fs.cd(kwargs["project_dir"]):
            platform = PlatformFactory.new(project_options["platform"])
            monitor_dir = os.path.join(platform.get_dir(), "monitor")
            device_helpers.register_filters(monitor_dir, options=kwargs)

    with fs.cd(kwargs["project_dir"]):
        monitor_dir = os.path.join(kwargs["project_dir"], "monitor")
        device_helpers.register_filters(monitor_dir, options=kwargs)
```

**In commands/device/helpers.py:**
Replace this:
```
def register_platform_filters(platform, options=None):
    monitor_dir = os.path.join(platform.get_dir(), "monitor")
    if not os.path.isdir(monitor_dir):
        return

    for name in os.listdir(monitor_dir):
        if not name.startswith("filter_") or not name.endswith(".py"):
            continue
        path = os.path.join(monitor_dir, name)
        if not os.path.isfile(path):
            continue
        load_monitor_filter(path, options)
```

With this:
```
def register_filters(monitor_dir, options=None):
    if not os.path.isdir(monitor_dir):
        return

    for name in os.listdir(monitor_dir):
        if not name.startswith("filter_") or not name.endswith(".py"):
            continue
        path = os.path.join(monitor_dir, name)
        if not os.path.isfile(path):
            continue
        load_monitor_filter(path, options)
```

WebDust21, 2023-08-30:
So this is a great feature I'm trying to leverage--but here almost 2 years after this issue was closed, I can't seem to figure out how to get it to work.  Been digging all over the place for answers, only finding one (unanswered) PIO forum post in the same vein (https://community.platformio.org/t/possible-to-write-serial-monitor-custom-filters-to-process-raw-binary-data/35190)

I'm trying to write a custom project-specific monitor filter.  Starting as simple as possible, I've started with https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor-custom-filters .
Copied the example code verbatim into a "demo.py" script, and put it in a folder in the immediate project directory, named "monitor".  Oh, but the docs indicate that "Each filter is a Python-based file and its name should have the filter_ prefix."  So the verbatim-copied Python script is located in "<project directory>/monitor/filter_demo.py".

Added the following to the platformio.ini "project environment" section:
```
monitor_speed = 19200
monitor_encoding = filter_demo
```

That didn't work.  I've tried "demo", "filter_demo", and "filter_demo.py" for "monitor_encoding" all with no luck.  I've also tried renaming the "demo" and "Demo" names inside "filter_demo.py" with "filter_demo" but to no avail.

So maybe it wasn't looking in the right place.  Based on https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/monitor_dir.html#projectconf-pio-monitor-dir I added what is indicated as the "default filter search path" to the head of the "platformio.ini" file:
```
[platformio]
monitor_dir = <Project>/monitor
```

Doesn't work.

Guessing based on the above page, I then tried this:
```
[platformio]
monitor_dir = monitor
```
This time it added a "\_pycache\_" folder inside the "monitor" directory.  But the "Serial Monitor" still failed.

All I ever get from the Serial Monitor is a long error trace, ending with, "LookupError: unknown encoding: _[whatever is on the 'monitor_encoding' line]_"

What am I missing here?

WebDust21, 2023-08-30:
Think I might have figured it out...
```
monitor_encoding = hexlify
```
works just fine--and I know that's a Python filter.  Which is what threw me off.

Apparently I missed the memo about using "monitor_filters"...

```
monitor_filters = filter_demo
```
surprisingly works.  Almost anticlimactic after all the errors...

commit 852c2523 (2021-11-05): Added support for custom device monitor filters // Resolve #3924