platformio/platformio-core #3925: Unit testing connecting string not displayed when not verbose

fionn-r, 2021-04-12:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.1

### Description of problem

When running platformio test, the `Connecting.....____......_____` doesn't appear unless verbose output is on.

Previously, it was shown all the time so I would know when to toggle the bootloader pin on my board.

#### Steps to Reproduce

1. Run `pio test` with a programmer attached (eg. ESP-Prog) but no microcontroller.
2. Run `pio test -v` or `pio test -v --without-building` and see the `Connecting...._____.......______` output as expected.

### Actual Results

```
pio test -e esp32 --upload-port COM10 --test-port COM10 --without-building
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing mocks in esp32 environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Uploading...
*** [upload] Error 2
=========================================================================================== [FAILED] Took 31.56 seconds ===========================================================================================
Test    Environment    Status    Duration
------  -------------  --------  ------------
mocks   esp32          FAILED    00:00:31.563
====================================================================================== 1 failed, 0 succeeded in 00:00:31.563 ======================================================================================
```

### Expected Results

```
pio test -e esp32 --upload-port COM10 --test-port COM10 --without-building -v
Collected 1 items

Processing mocks in esp32 environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Uploading...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
MethodWrapper(["upload"], [".pio\build\esp32\firmware.bin"])
Use manually specified: COM10
"c:\users\usr\.platformio\penv\scripts\python.exe" "C:\Users\usr\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM10" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 .pio\build\esp32\firmware.bin
esptool.py v3.0
Serial port COM10
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:79:4f:54
Uploading stub...
...
```

### Additional info

Let me know if there's anything else.

ivankravets, 2021-04-12:
Yes, we don't print build/upload output by default. You need to use `--verbose` flag.

closed 2021-04-12 by ivankravets

fionn-r, 2021-04-12:
Oh sorry, I thought I remembered that it used to and it was some sort of regression.

Thanks for making such a great tool!

ivankravets, 2021-04-12:
Happy coding with PlatformIO! 🚀