platformio/platformio-core #3926: Core: Network paths wrong in c_cpp_properties.json

Rotma, 2021-04-13:
**Operating system**: Win10
**PlatformIO Version** ) : 5.1.1

Network paths (UNC) in the "c_cpp_properties.json" file are missing the first '/'
Intellisense not working for the affected files...

project on network drive:  //nas/project  (or y:/project)  
Some libs on network     :  //nas/mylibs  (or y:/mylibs)

platformio.ini
================
[env:leonardo]
platform = atmelavr
board = leonardo
framework = arduino
lib_extra_dirs =\\nas\mylibs 

Tested also with "lib_extra_dirs =y:\mylibs"
same result wil also show up as /nas/mylibs

c_cpp_properties.json (the auto generated file)
=====================
// !!! WARNING !!! AUTO-GENERATED FILE!
// PLEASE DO NOT MODIFY IT AND USE "platformio.ini":
// https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
//
{
    "configurations": [
        {
            "name": "PlatformIO",
            "includePath": [
                "/nas/project/include",
                 ~~~~~~~~~~~~~~ (cannot find dir)
                "/nas/data/mylibs",
                ~~~~~~~~~~~~~~(cannot find dir)

JR

ivankravets, 2021-11-12:
Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2021-11-12 by ivankravets

Rotma, 2021-11-13:
Ivan,Thanks, it's working perfect now.Jan Rotmans

commit 4687665f (2021-11-12): Improved support for projects located on a network share // Resolve #3417 , Resolve #3926 , Resolve #4102