platformio/platformio-core #3927: Add Support for GigaDevice GD32F and GD32E chips

maxgerhardt, 2021-04-14:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

See https://www.gigadevice.com/products/microcontrollers/gd32/ and [catalog](https://www.gigadevice.com/wp-content/uploads/2021/03/2021-GigaDevice-Selection-Guide_Final.pdf) for GD32 ARM-type chips. GD32V (RISC-V based) chips are already supports, but not the GD32F (Cortex-M3 and Cortex-M4) and GD32E (Cortex-M23, Cortex-M33 and entry-level Cortex-M4). 

The chip vendor supplies its own STM32-like Simple Peripheral Library (SPL) framework as development tool, see http://www.gd32mcu.com/en/download/7?kw=

I currently support the GD32F130C6 and -C8 versions at https://github.com/maxgerhardt/pio-gd32f130c6 with the vendor's SPL pacakge, as `maxgerhardt/framework-spl@2.10301.0` (aka `GD32F1x0_Firmware_Library_v3.1.0` from vendor), and the GD32F103RC chip (with the same SPL version) under https://github.com/maxgerhardt/pio-gd32f103, but official support would be good.

I don't fully understand the chips but it might be that the GD32F103xx chips are exact clones of the STM32F103x chips, which also means they would implicitly be supported by the Arduino-STM32 core. For other chips like the GD32F130x series, I did not find a correlation.

maxgerhardt, 2021-04-20:
FYI: I've made an update to the `framework-spl` package to also incorporate the GD32F350xx series. 

The `framework-spl` package is at https://github.com/maxgerhardt/pio-framework-spl-with-gd32 with a new example (board definition + firmware code at) being at https://github.com/maxgerhardt/pio-gd32f350cb.

crosswick, 2021-04-22:
I would very much appreciate this - I'm looking into using platformio to program a GD32E103xx

maxgerhardt, 2021-06-18:
Update for those who are interested: I have recently gotten in contact with motivated people who also have connections to GigaDevice. They gave us a previously unreleased Arduino core for some GD32F3x chips. We are developing this further into fledged Arduino core with support for more boards at https://github.com/CommunityGD32Cores/GD32Core-New. Contributers welcome. 

For GD32 (ARM-type) chips I've also created https://github.com/maxgerhardt/platform-gd32 in which I plan a major overhaul and unification of all these smaller (SPL) projects linked above. 

Also note that mbed-os has support for some GigaDevices boards now.  See https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.15.7/targets/TARGET_GigaDevice for mbed-os 5.x and https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_GigaDevice for mbed-os 6.x. I plan to add mbed-os build support for these boards, too.

maxgerhardt, 2021-06-21:
@crosswick Just for my interest, what exact chip or board do you have in mind?

crosswick, 2021-06-21:
I'm working with the GD32E103C8T6 and GD32E103CBT6; boards are my own design and this one for prototyping: https://www.aliexpress.com/item/1005002023908240.html

maxgerhardt, 2021-06-21:
Yesterday I've added a board definition for the E103C8 [here](https://github.com/maxgerhardt/platform-gd32/tree/main/boards) and expanded the SPL blinky and UART example to work with it. If you clone [this](https://github.com/maxgerhardt/gd32-pio-projects), import the `gd32-spl-blinky` project folder in VSCode and change [these lines](https://github.com/maxgerhardt/gd32-pio-projects/blob/811ab334500a2b0462907cdb11fe395936da687c/gd32-spl-blinky/platformio.ini#L13-L14) to 

```ini
platform_packages = 
    framework-spl-gd32@https://github.com/maxgerhardt/pio-spl-gd32-new.git
```
can you do a successful build & upload for the `genericGD32E103C8` environment? The default upload protocol is `stlink` btw, but also `jlink` is supported. Since I don't have the board (yet) it would mean a lot to me if someone tested my platform :D

crosswick, 2021-06-21:
@maxgerhardt alright I've gotten it to build and upload, albeit with some RAM size-related warning, but no blinky. Perhaps I can show you the terminal output somewhere?

maxgerhardt, 2021-06-21:
Hm okay, can you open a new issue in https://github.com/maxgerhardt/platform-gd32/issues with the output and the used `platformio.ini`? Just to keep the discussion separate from here :)

maxgerhardt, 2021-07-28:
Update:

Me and a group of people have recently started a massive initiative to support GD32 chips in PlatformIO.

With
* new Arduino Core
* SPL support
*  all GD32 ARM type chips to be supported
  *  currently lots GD32F3, F1, F4, E1, E2 chips supported.

And we are also looking for contributors, Discord link is in the repo.

See the repos at
* https://github.com/CommunityGD32Cores/gd32-pio-projects
* https://github.com/CommunityGD32Cores/ArduinoCore-GD32
* https://github.com/CommunityGD32Cores/gd32-pio-spl-package
* https://github.com/CommunityGD32Cores/platform-gd32
* 
My smaller per-chip repos are considered outdated with the new repos above.

solawc, 2022-06-24:
Hello, I also have at least four gd32 development boards. I originally planned to build its own platform support package, but now it seems unnecessary. I will use and develop my own projects, such as grbl controller running on CNC or laser (my own ported version). I believe this is a very good experience.