platformio/platformio-core #3928: Unknown debug tool `um232h`

MainManu, 2021-04-15:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
      **Not sure if this fits here, but it fit even less into the other options.**

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.4

**PlatformIO Version** (`platformio --version`): 5.1.1

### Description of problem
For some reason the board listed in https://docs.platformio.org/en/latest/plus/debug-tools/um232h.html seems to be unknown to my plattformio version. 

#### Steps to Reproduce

1. add ```debug_tool = um232h``` to platformio.ini
2. try to upload code to a board

### Actual Results
Upload fails with message Unknown debug tool `um232h`

### Expected Results
1. the upload does not use the same board as the debugger, so it shouldn't be affected
2. the debugger listed in the docs should be known to plattformio

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32thing]
platform = espressif32
board = esp32thing
framework = arduino
lib_extra_dirs = /home/manuel/Arduino/libraries 
lib_deps = 
	sparkfun/SparkFun LSM9DS1 IMU @ ^2.0.0
	mbed-rohm/rohm-bh1790glc-driver@0.0.0+sha.4446bf1de21b

upload_port = /dev/ttyUSB1
monitor_speed = 115200 
debug_tool = um232h 
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info
I am uploading at /dev/ttyUSB0, the debugger is at /dev/ttyUSB1

MainManu, 2021-04-15:
99-platformio-udev.rules are already installed

valeros, 2021-04-19:
Hi @MainManu! According to the [docs](https://docs.platformio.org/en/latest/plus/debug-tools/um232h.html#platforms) this debug probe can only be used with the `GigaDevice GD32V` development platform.

closed 2021-04-19 by valeros