platformio/platformio-core #3933: \ wrongly escaped in openocd command

Windmill-City, 2021-04-21:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**:
Windows

**PlatformIO Version** (`platformio --version`):

### Description of problem
When using upload command like this:
```
upload_command = $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd -f $PROJECT_PACKAGES_DIR/tool-openocd/scripts/board/stm32f103zet6_warship.cfg -c "program $SOURCE 0x08000000 verify reset exit"
```
the `\` in the `$SOURCE` will treat as an escape char
then get this:
```
** Programming Started **
Error: couldn't open .piuildgenericSTM32F103ZE
                                              irmware.bin
** Programming Failed **
```

#### Steps to Reproduce

1.add above config to the platformio.ini
2.upload

### Actual Results
```
CURRENT: upload_protocol = custom
C:\Users\14491\.platformio\packages/tool-openocd/bin/openocd -f C:\Users\14491\.platformio\packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg -c "program .pio\build\genericSTM32F103ZE\firmware.bin 
0x08000000 verify reset exit"
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x06414041 (mfg: 0x020 (STMicroelectronics), part: 0x6414, ver: 0x0)
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x06414041 (mfg: 0x020 (STMicroelectronics), part: 0x6414, ver: 0x0)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08001d7c msp: 0x20010000
** Programming Started **
Error: couldn't open .piuildgenericSTM32F103ZE
                                              irmware.bin
** Programming Failed **
shutdown command invoked

*** [upload] Error 1
```

### Expected Results
```
CURRENT: upload_protocol = custom
C:\Users\14491\.platformio\packages/tool-openocd/bin/openocd -f C:\Users\14491\.platformio\packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg -c "program .pio/build/genericSTM32F103ZE/firmware.bin 
0x08000000 verify reset exit"
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x06414041 (mfg: 0x020 (STMicroelectronics), part: 0x6414, ver: 0x0)
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x06414041 (mfg: 0x020 (STMicroelectronics), part: 0x6414, ver: 0x0)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08001d7c msp: 0x20010000
** Programming Started **
Info : device id = 0x10036414
Info : flash size = 512kbytes
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x06414041 (mfg: 0x020 (STMicroelectronics), part: 0x6414, ver: 0x0)
shutdown command invoked
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:genericSTM32F103ZE]
platform = ststm32
board = genericSTM32F103ZE
framework = stm32cube
lib_deps = mincrmatt12/STM32Cube Middleware-FreeRTOS@10.0.1+f2-1.9.2.2
custom_freertos_config_location = Inc/FreeRTOSConfig.h
upload_protocol = custom
upload_command = $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd -f $PROJECT_PACKAGES_DIR/tool-openocd/scripts/board/stm32f103zet6_warship.cfg -c "program .pio/build/genericSTM32F103ZE/firmware.bin 0x08000000 verify reset exit"
debug_tool = custom
debug_server =
  $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd
  -f $PROJECT_PACKAGES_DIR/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
[platformio]
include_dir = Inc
src_dir = Sr
```

**Source file to reproduce issue:**
it has successful built

### Additional info

ivankravets, 2021-04-21:
See docs at https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-flags

You need to put 1 flag/option/value per 1 line for `upload_command` and `debug_server`

Windmill-City, 2021-04-25:
> See docs at https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-flags
> 
> You need to put 1 flag/option/value per 1 line for `upload_command` and `debug_server`

```ini
; Upload Configs
upload_protocol = custom
upload_command = $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd $UPLOAD_FLAGS
upload_flags =
  -f
  ${platformio.packages_dir}/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
  -c
  program $SOURCE $UPLOAD_OFFSET verify reset exit
```
Changed upload flags to this, but the `\` still wrongly escaped

commit 78182fea (2021-11-05): Disabled resolving of SCons variables when preprocessing "Interpolation of Values" // Resolve #3933