platformio/platformio-core #3934: Environment value didn't get replaced in debug_server

Windmill-City, 2021-04-21:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**:
Windows
**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.1.1
### Description of problem
I specipic debug_server like this
```
debug_server = 
    openocd 
    -f $PROJECT_PACKAGES_DIR/packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
```
and get this
```
embedded:startup.tcl:26: Error: Can't find $PROJECT_PACKAGES_DIR/packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
```
#### Steps to Reproduce

1. add above config line
2. run debug

### Actual Results
```
embedded:startup.tcl:26: Error: Can't find $PROJECT_PACKAGES_DIR/packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
```

### Expected Results
successful running debug

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:genericSTM32F103ZE]
platform = ststm32
board = genericSTM32F103ZE
framework = stm32cube
lib_deps = mincrmatt12/STM32Cube Middleware-FreeRTOS@10.0.1+f2-1.9.2.2
custom_freertos_config_location = Inc/FreeRTOSConfig.h
upload_protocol = custom
upload_command = $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd -f $PROJECT_PACKAGES_DIR/tool-openocd/scripts/board/stm32f103zet6_warship.cfg -c "program .pio/build/genericSTM32F103ZE/firmware.bin 0x08000000 verify reset exit"
debug_tool = custom
debug_server = 
    openocd 
    -f C:/Users/14491/.platformio/packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
[platformio]
include_dir = Inc
src_dir = Sr
```

**Source file to reproduce issue:**
it has successful built

### Additional info

ivankravets, 2021-04-24:
Could you try to replace `$PROJECT_PACKAGES_DIR` with `${platformio.packages_dir}`?  If it works for you, we will update our docs. See the other configuration variables https://docs.platformio.org/en/latest/projectconf/section_platformio.html

Windmill-City, 2021-04-25:
> Could you try to replace `$PROJECT_PACKAGES_DIR` with `${platformio.packages_dir}`? If it works for you, we will update our docs. See the other configuration variables https://docs.platformio.org/en/latest/projectconf/section_platformio.html

```
; Debug Configs
debug_tool = custom
debug_server = 
    openocd 
    -f
    ${platformio.packages_dir}/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
```
```
Processing genericSTM32F103ZE (platform: ststm32; board: genericSTM32F103ZE; framework: stm32cube)
--------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103ZE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103ZE (64k RAM. 512k Flash)
HARDWARE: STM32F103ZET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (custom) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-stm32cubef1 1.8.3 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
```

Windmill-City, 2021-04-25:
I success starting debugee by following
```ini
; Debug Configs
debug_tool = custom
debug_server = 
    openocd 
    -f
    ${platformio.core_dir}/packages/tool-openocd/scripts/board/stm32f103zet6_warship.cfg
```

commit 4839fe37 (2021-11-05): Improved PlatformIO directory interpolation (${platformio.***_dir}) in “platformio.ini” configuration file // Resolve #3934