platformio/platformio-core #3935: Industrial Shields Boards support

crmrosa, 2021-04-22:
- [ ] **Development Platform or Board**.

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10

**PlatformIO Version** (`platformio --version`):
Core 5.1.1·Home 3.3.4

### Description of problem
     I am looking for support in PlatformIO for Industrial Shields boards: http://apps.industrialshields.com/main/arduino/boards/package_industrialshields_index.json

#### Steps to Reproduce

1.
2.
3.

### Actual Results
No support found for these boards

### Expected Results
I expect to have support for these boards

alopez97, 2021-04-22:
Will be much apreciated if these boards are supported!

danielazaffalon, 2021-04-22:
I have the same issue, i appreciate some information about it! Thanks.

crmrosa, 2021-04-22:
Hi Daniel,

 

I am having some conversation with Industrial Shields teams.  I will keep you updated.

 

 

Claudio

Enviada em: quinta-feira, 22 de abril de 2021 09:37
Para: platformio/platformio-core ***@***.***>
Cc: Claudio Rosa ***@***.***>; Author ***@***.***>

GregVial, 2022-01-31:
I'm also interested!

gabrielfaleiro, 2022-03-09:
I'm also interested in this feature! May you need support on developing this feature? I'm willing to help anyhow.

johngunnarsson, 2022-08-11:
I'm also interested in this platoform!

KristoZ, 2023-01-20:
Bump almost two years later since the request - still looking forward to using Industrial Shields PLCs with PlatformIO. Apparently will have to fallback using the original Arduino extension for  Visual Studio Code.

KristoZ, 2023-02-10:
@MrThinkage Actually, no - as I was in a hurry I got by with the new Arduino IDE (2.0.3). It has the essential "Go to/Peek Definition" and was enough to rapidly integrate the Industrial Shields PLC within my project (two-way host communication, steppers, index switches, lights, temperature monitoring and control etc.).
Though I am still looking forward to have the PLC support in PlatformIO.

williamcoon, 2023-02-10:
We've started using the Industrial Shields PLC for automation projects and we are loving it but really having support in PlatformIO would make things way better. We have been relying on the Arduino extension in VSC and it's a headache. Any updates on progress on this?

Bart-van-Ingen, 2023-04-04:
Together with Max Gerhardt, (mostly Max to be fair) I was able to get a proof of concept to get the esp32 r19 plc to work with platformio! it still needs to get merged into the mainline branches of platformio, but for the rest it works just aswell as writing industrial shield compatible code in the arduino IDE.

The thread can be found here : https://community.platformio.org/t/adding-industrialshields-support-from-arduino-ide-to-platformio/32808

while this is currently only setup for the ESP32 boards, i believe it to be fairly trivial to get this to work for the atmelavr aswell!