platformio/platformio-core #3939: Serial Monitor should handle disconnects more gracefully

cinderblock, 2021-04-25:
### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.1.1`

### Description of problem
Using serial monitor on a USB based serial port crashes with Exception

#### Steps to Reproduce

1. Start a serial monitor to a device with a USB based serial port: `pio device monitor`
2. Reset the device

### Actual Results
```log
Exception in thread rx:
Traceback (most recent call last):
  File "C:\Users\camer\.platformio\python3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\camer\.platformio\python3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\camer\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "c:\users\camer\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
```

### Expected Results
An error message designed for a novice user to understand.

Ideally also automatic reconnect when the device comes back.

jcw, 2022-04-28:
For references, here is the slightly different traceback on MacOS:

```
--- exit ---
Traceback (most recent call last):
  File "/usr/local/Cellar/platformio/5.2.5/libexec/lib/python3.9/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/Cellar/platformio/5.2.5/libexec/lib/python3.9/site-packages/serial/tools/miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/usr/local/Cellar/platformio/5.2.5/libexec/lib/python3.9/site-packages/serial/serialposix.py", line 581, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured
```

belm0, 2022-05-15:
This is a basic thing (e.g. done by the Arduino IDE monitor), so I'd like to see it.

Most UI's would use some connection status indicator (like Arduino IDE uses backgroud color), but for a simple text terminal it should be enough to print `<disconnect>`.

belm0, 2022-05-15:
pyserial itself doesn't have any special reconnect support:  https://github.com/pyserial/pyserial/issues/558

if `monitor` implements this, it will have to be in a crude way like suppress the exception and silently retry connection every second

belm0, 2022-05-15:
`monitor` is using the "miniterm" tool of pyserial.  It's basically a stand-alone program, and exceptions happen it background threads.  So it will be hard to control error output.

options for implementing reconnect:
   1. live with the spamming on the output when there are disconnects, and during retry pollinge
   2. try to filter stderr to suppress the spam
   3. fork miniterm (about 1,000 loc) and enhance it for reconnect

belm0, 2022-05-15:
I implemented `--reconnect` option for pyserial miniterm... but then realized this project appears unmaintained (no commits in 6 months, backlog of PR's) 😢 

https://github.com/pyserial/pyserial/pull/651

ivankravets, 2022-05-15:
@belm0, thank you so much for your contribution to the OSS.

How about having a separate PR that just catches exceptions or exits gracefully without reconnecting? Typically, this issue arises when a user does not close the serial terminal but unplugs the board.

belm0, 2022-05-15:
I think a common case is uploading to a device.  It will reset the device and cause a lost connection.  This means that the monitor must be restarted after each upload.

Since I now have the enhanced miniterm written anyway, I'll try a PR that adds that code locally to platformio.

ivankravets, 2022-05-15:
You can't upload to the device if you don't close the serial monitor. Otherwise, you will receive `busy resources` error.

belm0, 2022-05-15:
I see-- it's my misunderstanding.  Uploading and monitoring have to be coordinated.

ivankravets, 2022-05-15:
Is it possible to refactor your PR to suppress unhandled exceptions?

jcw, 2022-05-15:
> Uploading and monitoring have to be coordinated.

It depends ... with ST-Link, a uC reset does not generate a USB reset/re-enumerate condition, so re-flashing will keep a terminal session intact if it's running in a separate window. I'm not sure this applies here, just wanted to flag this case.

egnor, 2022-06-07:
> I implemented `--reconnect` option for pyserial miniterm... but then realized this project appears unmaintained (no commits in 6 months, backlog of PR's) cry
> [pyserial/pyserial#651](https://github.com/pyserial/pyserial/pull/651)

:cry: my very _very_ humble outsider 2c would be-- miniterm is not magical, it's got a bunch of weird warts/bugs, and most of its features (menus that let you dynamically change settings and switch to new ports and such) aren't particularly interesting in this use case. Replacing it with something solid that just shuffles data between the serial port and the terminal (without blocking), handles disconnection gracefully, and allows robust cancellation would seem like a decent move to me.

ivankravets, 2022-06-07:
> Replacing it with something solid that just shuffles data between the serial port and the terminal

Agree with you. We will work on this.

belm0, 2022-06-07:
Agreed that miniterm is not magical.  It's just an example use of the pyserial library, and they added it to the public API for better or worse.

While it's a noble intention to replace it with something simple (and perhaps remove pyserial dependency altogether), it's not necessary.  There are probably more important places for platformio developers to invest their time.

Forking the miniterm example for our own purposes seems fine (as it uses the public pyserial API).  Ivan, please consider again this https://github.com/platformio/platformio-core/pull/4268#issuecomment-1126914452.

cinderblock, 2022-06-07:
@ivankravets The suggested "advanced" serial monitor leaves a *lot* to be desired. The main feature of `pio` automagically controlling the serial monitor during upload is sorely missing with 3rd party serial monitors.

ivankravets, 2022-06-11:
Hi All!

Great news! Huge refactoring to the device monitor logic and it seems this has been solved. Also, automatic reconnection has been implemented.

Please re-test with `pio upgrade --dev`.

Does it work as expected now?

closed 2022-06-11 by ivankravets

m1cr0lab, 2022-06-16:
@ivankravets Thank you very much for this correction!

zalexzperez, 2024-01-26:
> Does it work as expected now?

Hello,

I still have the issue on my ESP32-S3 using Windows 11 computer.
After resetting the board, I click on a blank space in the terminal and then press any keyboard key, this is shown:

```
Disconnected (ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22)))
Reconnecting to COM7     Connected!
```

I tried pio upgrade --dev, it did update but the problem is still there.

brentthierens, 2024-04-26:
Same problem here. When I reset the board, the output is just gone. It doesn't auto reconnect. Manually entering something will cause this message and the device works again. Also WIN11 if that matters.

mozzhead164, 2024-05-10:
Also same behaviour on an Adafruit MatrixPortal S3, if I use ESP.restart() in my script, the Serial Monitor disconnects when the ESP32S3 reboots. Also have to manually reset after uploading code for program to start. 
```
platform = espressif32
board = adafruit_matrixportal_esp32s3
framework = arduino
board_build.mcu = esp32s3
build_flags =
	-D ARDUINO_USB_CDC_ON_BOOT=1
monitor_speed = 115200
monitor_port = COM4
```

commit 7f351bc7 (2022-06-11): Automatically reconnect device monitor if a connection fails // Resolve #3939