platformio/platformio-core #3942: Pio check cppccheck addon error

Ian-Blockmans, 2021-04-30:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.1

### Description of problem

I am trying to use the cppcheck misra addon. If I run 

`pio check -v --flags "cppcheck: --addon=misra.json"`

I get a python error

#### Steps to Reproduce

1. get my project at https://github.com/Ian-Blockmans/efuse/tree/main/arduino/platformio%20mkrzero%20test
2. run `pio check -v --flags "cppcheck: --addon=misra.json"`

### Actual Results
```
Bailing out from checking F:\arduino\platformio mkrzero test\src\main.cpp since there was an internal error: Failed to execute 'c:\users\ian66\.platformio\penv\scripts\python.exe C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\misra.py --cli --rule-texts=misra-rules.txt "F:\arduino\platformio mkrzero test\src\main.cpp.dump"'. Traceback (most recent call last):

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\misra.py", line 3620, in <module>

main()

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\misra.py", line 3564, in main

checker.parseDump(item)

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\misra.py", line 3369, in parseDump

for cfgNumber, cfg in enumerate(data.iterconfigurations()):

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\cppcheckdata.py", line 927, in iterconfigurations

cfg.setIdMap(cfg_arguments)

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\cppcheckdata.py", line 740, in setIdMap

self.set_id_map(functions_arguments)

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\cppcheckdata.py", line 727, in set_id_map

scope.setId(IdMap)

  File "C:\Users\ian66\.platformio\packages\tool-cppcheck\addons\cppcheckdata.py", line 396, in setId

self.varlist.append(IdMap[v])

KeyError: '000001B540564840'
```

### Expected Results
The violations to the misra rules

boscs, 2021-08-25:
Hi,
I had the same issue in one of my own project with pio where I want static checks, so I might take a look to search for the underlying cause.
Did you find info on how to mitigate this or on what is causing it ? 

Thanks in advance,

valeros, 2021-09-10:
Hi @Ian-Blockmans ! I tried to reproduce the problem using your example, but it work just fine on my Win10. Anyway, I've just updated CppCheck to the latest v2.5 where this issue might be solved internally. It'd be great if you could update your PlatformIO installation to the latest from the development branch via the `pio upgrade --dev` command in your IDE terminal and try to run analysis again. 

Hi @boscs, any chance you could put together a minimal example to reproduce the issue?

boscs, 2021-09-12:
Hi @valeros,
My (short) investigation  has led me to believe that cppcheck is somehow mis-generating its 'dump' file that is passed to the add-ons. A var in the varlist part of the xml is created, but no associated token or variable is generated. This creates a key error when the add-on looks up  the var. 
I'll send you a sample project where you will be able to find an occurrence of this phenomenon in the main.cpp.dump file.
I haven't had the time to search more but maybe next week ;)

[template_project_for_modules.zip](https://github.com/platformio/platformio-core/files/7150132/template_project_for_modules.zip)

closed 2021-10-07 by valeros