platformio/platformio-core #3951: cppcheck code inspect fails on samd boards

sannoxdev, 2021-05-11:
### Configuration

**Operating system**:  Ubuntu 20.04
                                     5.4.0-73-generic #82-Ubuntu  x86_64 x

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.1

### Description of problem

When Inspect (pio check -v) is run against a SAMD based project the cppcheck code analysis fails with the following error:

```
Error: Found a breaking defect 'failed to expand 'SAMD_SERIES', Invalid ## usage when expanding 'part_is_defined'.' in /home/jimh/.platformio/packages/framework-cmsis-atmel/CMSIS/Device/ATMEL/sam.h:32
Please note: check results might not be valid!
Try adding --skip-packages

"severity=error<&PIO&>message=failed to expand 'SAMD_SERIES', Invalid ## usage when expanding 'part_is_defined'.<&PIO&>file=/home/jimh/.platformio/packages/framework-cmsis-atmel/CMSIS/Device/ATMEL/sam.h<&PIO&>line=32<&PIO&>column=0<&PIO&>callstack=[/home/jimh/.platformio/packages/framework-cmsis-atmel/CMSIS/Device/ATMEL/sam.h:32]<&PIO&>cwe=0<&PIO&>id=preprocessorErrorDirective"
```
Running check with --skip-packages works (pio check -v --skip-packages):

```
Component     HIGH    MEDIUM    LOW
-----------  ------  --------  -----
*/.pio         0        0        1
nofile         0        0        1
src            0        0        4

Total          0        0        6

Environment          Tool      Status    Duration
-------------------  --------  --------  ------------
uno                  cppcheck  PASSED    00:00:00.711
adafruit_feather_m0  cppcheck  PASSED    00:00:00.714
```
Attached zip file has a minimal project that is able to reproduce the problem. It has two environments setup; one for an UNO and one for a SAMD (Adafruit Feather M0). 

Uno env wil  pass cppcheck / SAMD fails.

#### Steps to Reproduce

1. Open attached project
2. Run Inspect or pio check
3. Error occurs on SAMD 

### Actual Results

See above

### Expected Results

Expect check using cppcheck to complete without failure

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
check_tool = cppcheck

[env:adafruit_feather_m0]
platform = atmelsam
board = adafruit_feather_m0
framework = arduino
check_tool = cppcheck
```

**Source file to reproduce issue:**
```
/**
 * Blink
 *
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */
#include "Arduino.h"

const int delayTime =1000;

// Set LED_BUILTIN if it is not defined by Arduino framework
// #define LED_BUILTIN 13

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(delayTime);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  delay(delayTime);
}
```

### Additional info

Test Project:

[Test2.tar.gz](https://github.com/platformio/platformio-core/files/6456127/Test2.tar.gz)

maxgerhardt, 2021-05-17:
Same as for https://community.platformio.org/t/code-inspection-not-returning-expected-result/21480/

maxgerhardt, 2021-05-23:
And also the same for any ESP8266 project

```` 
Error: Found a breaking defect ‘syntax error’ in %userprofile%.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\random.h:69
Please note: check results might not be valid!
```

jerabaul29, 2021-12-14:
Same issue here. Is there a way to go around this by adding some flags to platformio.ini or similar? :)

jerabaul29, 2021-12-14:
Btw, I can confirm that running in a platformio terminal:

```
$ pio check --skip-packages
```

works fine, but it seems to only check src, not the lib and include? Would be great to be able to check the src, lib, and include (guess we do not want to test the whole core that is provided by the board package, but these are where the potential "bad quality" user code live?).

red-scorp, 2023-08-01:
I assume this issue will be resolved with #4571

SpontaneousDuck, 2024-08-30:
Also seeing the error for a SADM21 board in PIO version 6.1.15. `check_skip_packages = yes` in the `platformio.ini` does allow me to inspect.

TimothyFran, 2025-08-31:
Any news on this?