platformio/platformio-core #3955: Scanning dependencies... Error: Could not find `ArduinoJson-esphomelib` dependency for `esphome-core` library

mglyman2018, 2021-05-15:
What kind of issue is this?

INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/nodemcuv2
Processing nodemcuv2 (platform: espressif8266@1.8.0; framework: arduino; board: nodemcuv2)
--------------------------------------------------------------------------------
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
LibraryManager: Installing v1.12.2
Downloading...
Unpacking  [####################################]  100%
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 25 compatible libraries
Scanning dependencies...
Error: Could not find `ArduinoJson-esphomelib` dependency for `esphome-core` library
========================== [ERROR] Took 15.21 seconds ==========================

I use docker to pull the latest esphome,and use vpn in case of not connecting to the internet,but it works like this.I use cp2102 nodemcu.

mglyman2018, 2021-05-15:
waiting for answer

ivankravets, 2021-09-13:
Please forward to https://community.platformio.org/

closed 2021-09-13 by ivankravets