platformio/platformio-core #3956: TEST Embedded Arduino Micro (ClearCommError)

ArthurREGNARD, 2021-05-18:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.1

### Description of problem
Got an error when testing on embedded Arduino Micro board. 

#### Steps to Reproduce
1. Upload the blink source code : No error and working
1. Source test code : https://docs.platformio.org/en/latest/tutorials/core/unit_testing_blink.html#tutorial-unit-testing-blink
1. run ``platformio test`

### Actual Results
```bash
$ platformio test -e micro
Verbose mode can be enabled via `-v, --verbose` option
Collected 3 items

Processing test_common in micro environment
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Uploading...

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file ".pio\build\micro\firmware.hex"
avrdude: writing flash (6016 bytes):

Writing | ################################################## | 100% 0.48s

avrdude: 6016 bytes of flash written
avrdude: verifying flash memory against .pio\build\micro\firmware.hex:
avrdude: load data flash data from input file .pio\build\micro\firmware.hex:
avrdude: input file .pio\build\micro\firmware.hex contains 6016 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.07s

avrdude: verifying ...
avrdude: 6016 bytes of flash verified

avrdude: safemode: Fuses OK (E:CB, H:D8, L:FF)

avrdude done.  Thank you.

Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

Error: Traceback (most recent call last):
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\platformio\commands\test\command.py", line 171, in cli
    "succeeded": tp.process(),
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\platformio\commands\test\embedded.py", line 52, in process
    return self.run()
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\platformio\commands\test\embedded.py", line 83, in run
    line = ser.readline().strip()
  File "c:\users\foxstraw\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 275, in read
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Le périphérique ne reconnaît pas la commande.', None, 22))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

### Expected Results
Test run on my Arduino Micro board.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:micro]
platform = atmelavr
board = micro
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <unity.h>

void test_led_builtin_pin_number(void)
{
    TEST_ASSERT_EQUAL(LED_BUILTIN, 13);
}

void test_led_state_high(void)
{
    digitalWrite(LED_BUILTIN, HIGH);
    TEST_ASSERT_EQUAL(digitalRead(LED_BUILTIN), HIGH);
}

void test_led_state_low(void)
{
    digitalWrite(LED_BUILTIN, LOW);
    TEST_ASSERT_EQUAL(digitalRead(LED_BUILTIN), LOW);
}

void setup()
{
    delay(2000);
    UNITY_BEGIN();
    RUN_TEST(test_led_builtin_pin_number);

    pinMode(LED_BUILTIN, OUTPUT);
}

uint8_t i = 0;
uint8_t max_blinks = 5;

void loop()
{
    if (i < max_blinks)
    {
        RUN_TEST(test_led_state_high);
        delay(500);
        RUN_TEST(test_led_state_low);
        delay(500);
        i++;
    }
    else if (i == max_blinks)
    {
        UNITY_END();
    }
}
```

### Additional info

maxgerhardt, 2021-05-18:
I think it has the same route cause as https://github.com/platformio/platformio-core/issues/3742.

ArthurREGNARD, 2021-05-19:
Got a temporary patch. 
Run the commande in two step: 
1. Build and Upload firmware. 
2. Testing firmware. 

Done with the commande : 
`platformio test -e micro -f test_embedded --without-testing && platformio test -e micro -f test_embedded --without-building --without-uploading`
Result :
```bash
$ platformio test -e micro --without-testing -f test_embedded && platformio test -e micro --without-building --without-uploading -f test_embedded
Verbose mode can be enabled via `-v, --verbose` option
Collected 3 items

Processing test_embedded in micro environment
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Uploading...

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file ".pio\build\micro\firmware.hex"
avrdude: writing flash (5028 bytes):

Writing | ################################################## | 100% 0.39s

avrdude: 5028 bytes of flash written
avrdude: verifying flash memory against .pio\build\micro\firmware.hex:
avrdude: load data flash data from input file .pio\build\micro\firmware.hex:
avrdude: input file .pio\build\micro\firmware.hex contains 5028 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.05s

avrdude: verifying ...
avrdude: 5028 bytes of flash verified

avrdude: safemode: Fuses OK (E:CB, H:D8, L:FF)

avrdude done.  Thank you.

======================================================================= [PASSED] Took 5.22 seconds =======================================================================
Verbose mode can be enabled via `-v, --verbose` option
Collected 3 items

Processing test_embedded in micro environment
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

ureToString     [PASSED]
-----------------------
1 Tests 0 Failures 0 Ignored
======================================================================= [PASSED] Took 5.68 seconds =======================================================================

Test           Environment    Status    Duration
-------------  -------------  --------  ------------
test_embedded  micro          PASSED    00:00:05.684
====================================================================== 1 succeeded in 00:00:05.684 ====================================================================== 
```

ivankravets, 2021-09-27:
Duplicate of https://github.com/platformio/platformio-core/issues/3742

closed 2021-09-27 by ivankravets