platformio/platformio-core #3957: mbed-os libraries have insufficient `library.json` generated, causing many to not work

maxgerhardt, 2021-05-18:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`): `version 5.2.0a6`

### Description of problem

Per [community topic](https://community.platformio.org/t/cannot-find-libraries-on-which-other-libraries-depend/21476/3?u=maxgerhardt) and following.

The mbed-os libraries available in the PlatformIO library registry have, when downloaded, as seemingly auto-generated `library.json` in them. Take e.g. the [X_NUCLEO_COMMON library](https://platformio.org/lib/show/8805/X_NUCLEO_COMMON).

The folder structure of this library is 

[image]

Notice that it has all the header files in individual subfolders.

The `library.json` of this library is

```json
{
    "authors": {
        "name": "ST",
        "url": "https://os.mbed.com/teams/ST/"
    },
    "description": "Generic mbed Extensions used by STM Expansion Board Firmware Packages.",
    "examples": [
        "https://os.mbed.com/users/dudmuck/code/X_NUCLEO_IKS01A1/",
        "<shortened as it's not important>"
    ],
    "frameworks": "mbed",
    "keywords": [
        "apis",
        "extension",
        "generic"
    ],
    "name": "X_NUCLEO_COMMON",
    "platforms": "*",
    "repository": {
        "type": "hg",
        "url": "https://os.mbed.com/teams/ST/code/X_NUCLEO_COMMON/"
    },
    "version": "0.0.0+sha.21096473f63e"
}
```

However, as you can see the `library.json` file does not have [build flags](https://docs.platformio.org/en/latest/librarymanager/config.html#flags) option to include those subdirectories in which the actual library headers are. 

Other libraries refer to this library as 

```
#include "DevI2C.h"
```
(see e.g. [HTS221 library](https://platformio.org/lib/show/3717/HTS221/examples)) and thus will throw an error during compilation abotu the file not being found.

[image]

#### Steps to Reproduce

1. Use provided `platformio.ini` and code from below
2. Attempt compilation

### Actual Results

Compilation failure, header file not found.

### Expected Results

The `library.json` correctly contains the build flags options to include all code subdirectories (while of course still ignoring directories like `test/`) so that the libraries header files are correctly found.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a 
framework = mbed
lib_ldf_mode = chain+
lib_deps = 
    mbed-mqtt/MQTT@0.0.0+sha.9cff7b6bbd01
    mbed-st/HTS221@0.0.0+sha.ccf7f36492ae
    mbed-farnell24-iot-team/wifi-ism43362@0.0.0+sha.766454e296c3
```

**Source file to reproduce issue:**
```cpp
#include <mbed.h>
#include "ISM43362Interface.h"
#include "TCPSocket.h"
#include "MQTTmbed.h"
#include "MQTTClient.h"
#include "HTS221Sensor.h"

int main() {

  // put your setup code here, to run once:

  while(1) {
    // put your main code here, to run repeatedly:
  }
}
```

### Additional info

Manually fixed `platformio.ini`

```ini
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a 
framework = mbed
lib_ldf_mode = chain+
lib_deps = 
    mbed-mqtt/MQTT@0.0.0+sha.9cff7b6bbd01
    mbed-st/HTS221@0.0.0+sha.ccf7f36492ae
    mbed-farnell24-iot-team/wifi-ism43362@0.0.0+sha.766454e296c3
build_flags = 
    -D"wait_ms=ThisThread::sleep_for"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/FP"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/MQTT"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/ST_INTERFACES/Actuators"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/ST_INTERFACES/Common"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/ST_INTERFACES/Communications"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/ST_INTERFACES/Sensors"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/wifi-ism43362/ISM43362"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/wifi-ism43362/ISM43362/ATParser"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/wifi-ism43362/ISM43362/ATParser/BufferedSpi"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/wifi-ism43362/ISM43362/ATParser/BufferedSpi/Buffer"    
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/X_NUCLEO_COMMON/DbgMCU"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/X_NUCLEO_COMMON/DevI2C"
    -I"$PROJECT_LIBDEPS_DIR/$PIOENV/X_NUCLEO_COMMON/DevSPI"
```

ivankravets, 2021-09-26:
Thanks for the report! It was a strange issue. We re-packed all mbed libraries. Please re-test with `pio upgrade --dev`, remove `.pio` folder, and `pio system prune -f` (or remove `.cache` folder from `.platformio` folder).

closed 2021-09-26 by ivankravets

maxgerhardt, 2021-09-26:
Updated, pruned and retested with

```ini
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a 
framework = mbed
lib_ldf_mode = chain+
lib_deps = 
    mbed-mqtt/MQTT@0.0.0+sha.9cff7b6bbd01
    mbed-st/HTS221@0.0.0+sha.ccf7f36492ae
    mbed-farnell24-iot-team/wifi-ism43362@0.0.0+sha.766454e296c3
build_flags = 
    -D"wait_ms=ThisThread::sleep_for"
```
and code above, compiles without any problems now.

commit b6f78367 (2021-09-26): Allowed to override a default library builder via a new ``builder`` field in a ``build`` group of `library.json` // Resolve #3957