platformio/platformio-core #3959: PlatformIO Core does not work on msys 2

Haringat, 2021-05-19:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): probably latest since just installed via pip ("platformio --version" does not work for reasons below)

### Description of problem

Platformio is not usable on msys2. It shows the same error output regardless of what you pass in:

#### Steps to Reproduce

1. Install msys2
2. Install python and pip inside msys
3. Install platformio using pip
4. Run platformio
5. Watch the fireworks

### Actual Results

```
$ platformio -h
don't know how to enumerate ttys on this system.
! I you know how the serial ports are named send this information to
! the author of this module:

sys.platform = 'msys'
os.name = 'posix'
pySerial version = 3.5

also add the naming scheme of the serial ports and with a bit luck you can get
this module running...
Usage: platformio [OPTIONS] COMMAND [ARGS]...
Try 'platformio -h' for help.

Error: No such command "ci"
```

### Expected Results

*usage*

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
irrelevant

**Source file to reproduce issue:**
irrelevant

### Additional info

ustsv, 2021-09-15:
After patch **click** it work.  [Use in MSYS2 w/o termios available](https://github.com/pallets/click/pull/1666)
and platformio/builder/tools/piomaxlen.py need restore version @5.0.4. In version 5.1.0, 5.2.0 fails:
`Using tempfile C:/tmp/Marlin/.pio/build/morpheus_ustsv/tmpthqhsc3m.tmp for command line:<skip...>Invalid switch - "tmp".`

ivankravets, 2021-09-16:
Why did you close this PR https://github.com/pallets/click/pull/1666 ?

ustsv, 2021-09-17:
They said that the problem is in implementing MSYS2. Not their problem.