platformio/platformio-core #3963: Linux incorrect project folder path generation

TaremAkabekov, 2021-06-04:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.4 (Python 3.8.5 & 3.9)

**PlatformIO Version** (`platformio --version`): 
5.2.0a6, 
Core 5.1.1
Home 3.3.4

### Description of problem

I am currently facing a problem regarding project generation when working with core & home on ubuntu 20.4.
When creating a project from within the PlatformIO Home web interface all files will be created into the predefined path. But instead of forward slashes “/” backslashes “” are used to divide the path. This obviously doesnt work on Linux systems. The project structure than is generated with the backslashes on my root directy (e.g. “/data\downloads\test_platform_io_esp_32_project”).
The issue is occuring when I create a project (and potentially when loading it as well) not when I add a new file. 

I have installed Platform IO twice.
Once with pip install platformio and the other time with the recommended way Installer Script
Both end up with the same result.

Refering to [platformio-community: linux-incorrect-project-folder-path-generation](https://community.platformio.org/t/linux-incorrect-project-folder-path-generation/21784/5)
#### Steps to Reproduce

0. At the beginning the root-directory (":/") is not containing any platform-io related folder:
```
:/$ ls
bin   cdrom  dev  home  lib32  libx32      media  opt   root  sbin  snap  swap.img  tmp  var
boot  data   etc  lib   lib64  lost+found  mnt    proc  run srv   sys       usr
```

1. Starting the server with the activated [Installer Script](https://docs.platformio.org/en/latest/core/installation.html#id28) virtualenv in the folder data/platform-io-testing:
```  (penv) data/platform-io-testing:/$ pio home --host 127.0.0.1 --port 800```

 2. open localhost:800/ and create a new project with the name "new_test_project" in the folder "data/platform-io-testing"

 3. It gets created and I recieve the message: 
```
Project has been successfully initialized
Board: esp32doit-devkit-v1, framework: arduino, location: /data\platform-io-testing\new_test_project
```
### Actual Results

Which is correct. When I now list the content of the root directory I find the directory:

```
 bin     data                      dev    lib     libx32       mnt    root   swap.img   usr
 boot   'data\platform-io-testing\new_test_project'   etc    lib32   lost+found   opt    run    snap                        sys        var
 cdrom  home   lib64   media        proc   sbin   srv                         tmp
```

 But the folder is no where near to be found in /data/platform-io-testing/new_test_project
If I list the content of this incorrect path all the jucy platform-io files can befound (platformio.ini etc.)

### Expected Results
I expect to find the created project in `/data/platform-io-testing/new_test_project`

### Additional info

To make reproducing the problem easier, I have created a docker container (heavily inspired by sglahn:https://github.com/sglahn/docker-platformio-core/blob/master/Dockerfile)
```
FROM python:3.8.5-slim

USER root

RUN pip install -U platformio==5.1.1 && \
    mkdir -p /workspace && \
    mkdir -p /.platformio && \
    chmod a+rwx /.platformio && \
    apt update && \
    apt install -y git && \
    apt-get clean autoclean && \
    apt-get autoremove --yes && \
    rm -rf /var/lib/{apt,dpkg,cache,log}/
    
USER 1000
WORKDIR /workspace
EXPOSE 800

ENTRYPOINT ["pio", "home", "--host", "0.0.0.0", "--port", "800"]
```
```
docker build -t platform-io .
docker run -p 8403:800 --name platform-io -v %LOCAL_PATH_TO_PROJECTS%:/Documents/PlatformIO/Projects -d -it platform-io 
docker logs -f platform-io
```

valeros, 2021-09-29:
Hi @TaremAkabekov ! It's strange, but I cannot reproduce the issue even with your Dockerfile (although I needed to remove the `USER 1000` line, otherwise I got a `Permission Denied` error when creating a new project):

[image]

As a side node, I'm not sure if it's a typo, but I noticed you're using a Windows-like env variable substitution symbols `%LOCAL_PATH_TO_PROJECTS%`. Any chance you're using the Windows Subsystem for Linux?

cbundy, 2022-03-18:
@TaremAkabekov were you viewing this remotely?

I am running `pio home` on a linux machine (not WSL), then accessing the webpage from a Windows machine and I'm seeing the same behaviour. 

**Problem appears to be in the Platform IO Home project:** 

The front end defines a [variable called `IS_WINDOWS`](https://github.com/platformio/platformio-home/blob/1055b98034feb6e1e9f4910462c24e1e957c31c0/app/config.js#L17) that detects whether the browser is running on Windows, the display and also [the requested paths are then altered in a few places](https://github.com/platformio/platformio-home/search?q=IS_WINDOWS) based on this variable

I suppose the assumption here was that the platform PIO Core is running on will be the same as where the user is. This doesn't really work for remote development.

I'm using VS Code Remote SSH from Windows, connecting to VS Code on my ARM Linux machine, everything works quite well with the exception of this so far.

skullcrushercmd, 2024-04-07:
i also get this issue, im using windows but remote ssh to my raspberry pi to use platformio, trying to change the directory name but unsuccessful.

20Lush, 2024-04-21:
> i also get this issue, im using windows but remote ssh to my raspberry pi to use platformio, trying to change the directory name but unsuccessful.

Exact same situation - VSCode Remote SSH into a raspberry pi 5, `pio home` in the pio terminal, and using either the web GUI or the in-IDE GUI to create a project it produces a "Could not find package" error but I suspect the actual issue is:
[image]
Where the postfix of the file path is Windows-style rather than linux compliant

skullcrushercmd, 2024-04-23:
> Exact same situation - VSCode Remote SSH into a raspberry pi 5, `pio home` in the pio terminal, and using either the web GUI or the in-IDE GUI to create a project it produces a "Could not find [...]

Yeah I noticed that, kina gave up on doin it cuz it was slow anyways

MojitoShader, 2024-07-12:
I have the same Problem with WSL2 and PIO. I just rename the Projects under Inspect and then it works fine.
Or using the Windof explorer to navigate into the Project directory and change it there.. that works but is a Pain.

brownbrowny, 2025-03-06:
Some colleagues and me use WSL2 + PIO and encounter the same problem. It is easily solved by manually renaming the folder but it's extra work, especially when you only want to test something quickly.

ShuhuaGao, 2025-07-13:
Same problem on WSL2.

gwilherm, 2025-10-04:
Same issue with platformio-core 6.1.18-1 on Manjaro Linux with Firefox:
Firefox user-agent is (dunno why) : "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/128.0"