platformio/platformio-core #3965: Output from custom target *very* slow to appear in the terminal

adbancroft, 2021-06-06:
### Goal 
Add a custom target that launches a simulator: simavr in my case.

### Configuration

**Operating system**: Window 10 64 bit Version 1909

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.0a6

### Description of problem

Simulator output is incredibly slow to echo in the terminal: up to the point it looks like nothing is happening. If I wait long enough (minutes) it eventually appears. 

Running simavr on it's own produces output line by line.

#### Steps to Reproduce

1. Setup a project using the ini, cpp & python script below.
2. Build
3. Launch the 'Simulate' custom target. E.g. `pio run -t Simulate`

### Actual Results

Simulator output doesn't appear in the terminal

### Expected Results

Simulator output (to stdout & stderr) is echoed in the terminal as soon as the child simavr process prints to stdout/stderr.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
extra_scripts = simavr_env.py
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  Serial.println("TESTING");
}

void loop() {
  if (counter % 10 == 0) {
    Serial.println(counter); 
  } else {
    Serial.print(counter); Serial.print("-");
  }
  counter++;
  delay(100);
}
```

### Additional info

[Custom python script](https://github.com/adbancroft/platformio/blob/main/simavr_env.py)

valeros, 2021-09-24:
Hi @adbancroft ! It seems your extra script is no longer available. Could you please upload it somewhere so I can reproduce the issue?

adbancroft, 2021-09-24:
@valeros Original link updated.

Also here: https://github.com/ridiculousfish/libdivide/blob/master/test/avr/simavr_env.py

ivankravets, 2022-05-04:
Duplicate of https://github.com/buserror/simavr/issues/489

closed 2022-05-04 by ivankravets