platformio/platformio-core #3970: LDF doesn't detect src dir of local library

fazerlicourice7, 2021-06-12:
- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Mac OS Big Sur

**PlatformIO Version** (`platformio --version`): 5.1.1

### Description of problem

LDF fails to detect the `src` in a library causing some symbols to be undeclared. You can see that LDF doesn't identify the src dir in the snippet below:

Snippet of the output of `pio run -e <env> -t upload -v`
```
Dependency Graph
|-- <include> (/Users/vamshi/Documents/tests-verification/port/Middlewares/Third_Party/FreeRTOS/include)
|   |-- <src> (/Users/vamshi/Documents/tests-verification/port/Middlewares/Third_Party/FreeRTOS/src)
|-- <include> (/Users/vamshi/Documents/tests-verification/port/Middlewares/Third_Party/LwIP/include)
|   |-- <src> (/Users/vamshi/Documents/tests-verification/port/Middlewares/Third_Party/LwIP/src)
|   |   |-- <include> (/Users/vamshi/Documents/tests-verification/port/Middlewares/Third_Party/FreeRTOS/include)
|   |   |   |-- <src> (/Users/vamshi/Documents/tests-verification/port/Middlewares/Third_Party/FreeRTOS/src)
|-- <include> (/Users/vamshi/Documents/tests-verification/port/Drivers/STM32F7xx_HAL_Driver/include)
Building in release mode
arm-none-eabi-gcc 
```

Directory tree: 
[image]

#### Steps to Reproduce

1. Create new workspace for an stm32 board using the CMSIS framework and `ststm32` platform
2. Copy the STM32xxxx HAL into the library folder
3. reorganize HAL into `include` and `src` folders.
4. add `library.json` (optional)
5. Write test code with `#include "stm32f7xx_hal.h"`, and use various HAL functions, for example `HAL_Init()`.
6. Add build flag `DUSE_HAL_DRIVER`
6. try to build with `pio run`

### Actual Results

```
 undefined reference to `HAL_GPIO_WritePin'
 undefined reference to `HAL_Delay'
 undefined reference to `HAL_GetTick'
 undefined reference to `HAL_GPIO_Init
 ...
```

### Expected Results

Should build fine.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[nucleo_f767zi]
platform = ststm32
board = nucleo_f767zi
framework = cmsis ; stm32cube
board_build.stm32cube.custom_config_header = yes
lib_extra_dirs = Drivers/STM32F7xx_HAL_Driver, Middlewares/Third_Party/LwIP, Middlewares/Third_Party/FreeRTOS
lib_ldf_mode = chain

[platformio]
include_dir = Inc
src_dir = Src

[env:port]
extends = nucleo_f767zi
src_filter = -<*> +<port>
build_flags = -IDrivers/STM32F7xx_HAL_Driver/src -IDrivers/STM32F7xx_HAL_Driver/include -IInc -IInc/port -mcpu=cortex-m7 -mthumb -mfloat-abi=softfp -DUSE_HAL_DRIVER
```

**Source file to reproduce issue:**
```cpp
#include "stm32f7xx_hal.h"
int main(void)
{
  
  HAL_Init();

  SystemClock_Config();
}

void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 216;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Activate the Over-Drive mode
  */
  if (HAL_PWREx_EnableOverDrive() != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_7) != HAL_OK)
  {
    Error_Handler();
  }
  PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_I2C1|RCC_PERIPHCLK_I2C2;
  PeriphClkInitStruct.I2c1ClockSelection = RCC_I2C1CLKSOURCE_PCLK1;
  PeriphClkInitStruct.I2c2ClockSelection = RCC_I2C2CLKSOURCE_PCLK1;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
}

```

### Additional info

My project also uses `LwIP` and `FreeRTOS`.

fazerlicourice7, 2021-06-12:
NVM. Changing 
```
lib_extra_dirs = Drivers/STM32F7xx_HAL_Driver, Middleware...
```

to 

```
lib_extra_dirs = Drivers, Middleware...
```

fixed it. Not sure why, since the other dirs provided there go to the same depth. ie, the sub-dirs are `src` and `inc`.

fazerlicourice7, 2021-06-18:
Scratch that. That did not fix it!

fazerlicourice7, 2021-06-18:
Explicitly doing `build_flags = -IDrivers/STM32F7xx_HAL_Driver/src -IDrivers/STM32F7xx_HAL_Driver/Inc` fixed it.
But I don't think that really counts as fixing it.

ivankravets, 2021-09-13:
Please forward to https://community.platformio.org/

closed 2021-09-13 by ivankravets