platformio/platformio-core #3977: i have this problem when compile esp3d v 3

ranginak, 2021-06-15:
> Executing task: C:\Users\saeed\.platformio\penv\Scripts\platformio.exe run --environment esp8266dev <

Processing esp8266dev (platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/platformio/platform-espressif8266.git
Error: Traceback (most recent call last):
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 70, in process
    p = PlatformFactory.new(self.options["platform"])
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 54, in new
    raise UnknownPlatform(pkg_or_spec)
platformio.platform.exception.UnknownPlatform: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 133, in cli
    process_env(
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 72, in process
    self.cmd_ctx.invoke(
  File "c:\users\saeed\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 325, in platform_install
    pkg = pm.install(
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 51, in install
    pkg = super(PlatformPackageManager, self).install(
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 54, in new
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\saeed\.platformio\penv\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\saeed\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\saeed\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

[... 68 lines cut ...]
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\saeed\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\saeed\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "C:\Users\saeed\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp8266dev'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

ranginak, 2021-06-15:
i have the latest versin

maxgerhardt, 2021-06-19:
Do you have [git](https://git-scm.com/downloads) installed?

ranginak, 2021-06-21:
> Do you have [git](https://git-scm.com/downloads) installed?

thanks the problem solved.

closed 2021-06-24 by valeros