platformio/platformio-core #3978: nRF5340-DK support

dovoto, 2021-06-17:
What kind of issue is this?

[X ] **Feature Request**.

Add support for nRF53 platform (specifically nRF5340-DK board).

      This board is similar enough to nrf52 that it might be accomplished with a board definition vice a new platform but it is a dual processor system so maybe not...

Working on this myself now but starting from a very ignorant place.

antoniomolram, 2022-10-18:
any update on this? It is almost one year from last update

jonnor, 2023-04-18:
One would hope this would come from having Zephyr support. But NRF53 needs a recent Zephyr, while it seems PlatformIO is stuck on Zephyr 2.x: https://github.com/platformio/platform-nordicnrf52/issues/141

kokroo, 2023-10-13:
> One would hope this would come from having Zephyr support. But NRF53 needs a recent Zephyr, while it seems PlatformIO is stuck on Zephyr 2.x: [platformio/platform-nordicnrf52#141](https://github.com/platformio/platform-nordicnrf52/issues/141)

Platform IO has Zephyr 3.x now.

rajeev1986, 2024-04-02:
Yeah its dead in the water. Once I reached out to the folks at PIO and I was asked if I need support for nRF91 or latest devices the price tag was ~10-15k for development support. 
Right away from that day I am fully on nRF Connect SDK. 

My understanding is that series before nRF52 were all using Arduino like programing structure/core beneath so PIO was easily able to use that / hijack that but now more work is involved so no one is making a move.