platformio/platformio-core #3981: couldn't install platform (advanced installation)

yutech23, 2021-06-22:
PS C:\Users\YuTech> pio platform install https://github.com/deneyapkart/deneyapkart-platformio-core.git
Platform Manager: Installing git+https://github.com/deneyapkart/deneyapkart-platformio-core.git
Error: Traceback (most recent call last):
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\yutech\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\yutech\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\yutech\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 325, in platform_install
    pkg = pm.install(
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 51, in install
    pkg = super(PlatformPackageManager, self).install(
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 54, in new
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\yutech\.platformio\penv\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\YuTech\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\YuTech\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Sistem belirtilen dosyayı bulamıyor

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
[image]

My system : 
Windows 10 Home 20H2, 
Visual Studio Code 1.57.1
PlatformIO 2.3.2
Python 3.9.5

valeros, 2021-06-23:
Hi @yutech23, you need to install a Git client https://git-scm.com/downloads

closed 2021-06-23 by valeros

yutech23, 2021-06-23:
Thank you so much. My problem is done.