platformio/platformio-core #3984: tool-CPPCHECK for aarch64

GiulioCocconi, 2021-06-23:
What kind of issue is this?

- [ x ] **Development Platform or Board**.

------------------------------------------------------------------

### Configuration

**Operating system**: Raspberry OS 64bit

**PlatformIO Version** (`platformio --version`): 5.2.0a7

### Description of problem
I can't use the check code function on my Raspberry PI 4 because tool-cppcheck for aarch64 doesn't exists

#### Steps to Reproduce

1. Open PIO Home
2. Inspect a Project with cppcheck

### Actual Results

```
Error: Tool Manager: Installing platformio/tool-cppcheck @ ~1.241.0

Error: Could not find the package with 'platformio/tool-cppcheck @ ~1.241.0' requirements for your system 'linux_aarch64'
Error: Checking megaatmega2560 > cppcheck (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-cppcheck @ ~1.241.0

Error: Could not find the package with 'platformio/tool-cppcheck @ ~1.241.0' requirements for your system 'linux_aarch64'

```
### Expected Results
PIO checks the code

maxgerhardt, 2021-06-24:
Does it work if you add

```
platform_packages =
   tool-cppcheck@https://github.com/maxgerhardt/tool-cppcheck-arm64.git
```
in the `platformio.ini`?

GiulioCocconi, 2021-06-25:
No...
`git version 2.23.0
Cloning into '/home/pi/.platformio/.cache/tmp/pkg-installing-n17f3vc_'...
remote: Enumerating objects: 66, done.
remote: Counting objects: 100% (66/66), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 66 (delta 23), reused 65 (delta 23), pack-reused 0
Unpacking objects: 100% (66/66), done.
Tool Manager: Installing platformio/tool-cppcheck @ ~1.241.0
Error: Could not find the package with 'platformio/tool-cppcheck @ ~1.241.0' requirements for your system 'linux_aarch64'`

GiulioCocconi, 2021-06-25:
Also, if I manually clone the repository and try to use the executable, this error comes out...
`
./cppcheck: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ./cppcheck)
./cppcheck: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./cppcheck)

`

maxgerhardt, 2021-06-25:
Ah too bad, then the binaries https://ubuntu.pkgs.org/20.04/ubuntu-universe-arm64/cppcheck_1.90-4build1_arm64.deb.html are not working for your operating system.. I guess you are running Rasbian? What does `uname -a` and `lsb_release -a` say?

GiulioCocconi, 2021-06-26:
Uname: `Linux nimbus2000 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux`
Lsb: `No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster`

valeros, 2021-09-24:
Hi @GiulioCocconi ! The Cppcheck package for aarch64 is now available in the PlatformIO registry, please try to run analysis on your project again.

closed 2021-09-24 by valeros