platformio/platformio-core #3985: ...[WinError 2]...

PlatformIO-SoeN, 2021-06-24:
I use PlatformIO without problems until the update to PlatformIO IDE 2.3.2
Now I have the same problems ...[WinError 2]....
I Try to:
1. Uninstall PlatformIO from VsCode, uninstall VsCode, uninstall Python
Restart my PC, and start new installation depend on "https://platformio.org/"
again the same problems ...[WinError 2]....

2. Uninstall PlatformIO from VsCode, uninstall VsCode, uninstall Python
Restart my PC, deactivate my Kaspersky-Anti-Virus and start new installation depend on "https://platformio.org/"
again the same problems ...[WinError 2]....

3.
I tried a number of suggested solutions to this problem and also tried older versions of VsCode and PlatformIO IDE
again the same problems ...[WinError 2]....

I have no idea how I should get on there.
I do not understand that this problem is not easy to recreate by the developers of platformIO!

I've really liked the PlatformIO IDE so far

_Originally posted by @PlatformIO-SoeN in https://github.com/platformio/platformio-core/issues/2321#issuecomment-867540095_

TR7Mike, 2021-09-10:
Originally posted [here](https://github.com/platformio/platformio-core/issues/2321?utm_source=platformio&utm_medium=piohome#issuecomment-916937667), but it looks like the issue is closed.

Just installed PlatforIO and it's now showing as Core 5.2.0b1 / Home 3.3.4 and tried to add a library to my existing project.

Error shows:
Libraries: Installed - PIO Core Call Error: "[WinError 2] The system cannot find the file specified"

Can't see anything of reference in the log files at all. Library installed tab just has spinning waiting icons.

Path to project is "E:\Documents\Arduino\Projects\MyProjectName"

ivankravets, 2021-09-13:
Please forward to https://community.platformio.org/

closed 2021-09-13 by ivankravets